EdgeTX/edgetx #1828: Play Track (Special Function) being active prevents the radio from shutting down

MrTucks, 2022-04-14:
**Describe the bug**
If the Special Function "Play Track" is currently playing a track, the radio can't shut down. 
I know, for longer audio files like music the BgMusic Special Function is better suited (and doesn't have this bug) but I still think that this is a bug. But I totally understand if it gets filed unter "wont fix. Just use BgMusic". 
**To Reproduce**
Create a special function "Play Track" and select a longer music track. I used a song that's a few minutes long.
Activate the special function (I did using a switch).
Try to shut down the radio.
Sleeping face appears on-screen, but music keeps playing.
After a few seconds the speaker clicks, and I'm back on the main screen of the radio, for a few seconds the battery voltage reads 0.2V. If the switch was flipped back, the music stops playing, if the switch is still active, it restarts.

**Expected behavior**
I expected the same behavior as with the BgMusic special function, which is:
Shut down the radio. Music stops playing. Radio shuts down.

**General information**
 - EdgeTX version: 2.6.0
 - Hardware: FrSky Taranis QX7 ACCST

pfeerick, 2022-04-15:
*Just use BgMusic* 😝 

But seriously, that does sound like a bug... I don't think audio playback should interfere with shutdown.

raphaelcoeffic, 2023-09-05:
@pfeerick is it still a thing?