EdgeTX/edgetx #1839: Possible Bug/User Error NV14 Using 2.6 companion  resets radio calibration

davidabevan, 2022-04-16:
Hello
Nirvana Nv14 just flashed to v2.6rc 
I have just pulled my old nirvana nv14 out to experiment with the new edgetx. All is ok so far except I am experiencing my radio stick calibration, battery meter calibration and low battery warnings they appear to get reset after re writing models back to radio using the edgetx companion.
Basically what i did was in companion pull all my models then created new page and dragged the required models back over in the order I wanted. all the model data seems fine but I had to go through the radio settings and calibrate again.
I really dont know if user error or a bug so letting you know.
Thanks for giving the nv14 a new lease of life and finally companion.
Dave B

pfeerick, 2022-04-16:
First order of business will be to jump to 2.7.0, as it is possible it was a bug that was fixed during the 2.7 development cycle. Make sure you run *both* EdgeTX 2.7.0 Companion and firmware, and see if this happens again. After that, it will require some fiddling around to see if we can reproduce this and find out what the cause is. :)

davidabevan, 2022-04-16:
Yes no problems thanks for replying i can try this tonight and let you
know..

davidabevan, 2022-04-16:
Ok
2.7 seems to resolve this issue for me, Thanks

closed 2022-04-16 by davidabevan