EdgeTX/edgetx #1840: Add more than five options to widgets

kristjanbjarni, 2022-04-16:
Creating complex widgets scripts is very limited currently because of only having maximum of five available options for customizing widgets. I would like to see this maximum raised to allow for more options.

pfeerick, 2022-04-17:
I would have sworn there was an issue for this already, but it seems like there isn't.  In my own experience I already hit the limit of options for internal widgets so couldn't implement https://github.com/EdgeTX/edgetx/issues/1095 yet... so wanted to improve that. I'm thinking the simplest way is to make it a fully-screen window, this allow it to scroll normally and such... as the current popup has issues, and really isn't necessary as it's not like it can update dynamically as you change settings.

kristjanbjarni, 2022-04-17:
Yes, full screen makes much more sense. This is a really limiting factor for creating useful and easy to use widgets.

Eldenroot, 2022-07-05:
https://github.com/opentx/opentx/pull/8052 maybe these too?

pfeerick, 2022-07-05:
Er... 1x2, 1x3, 1x4 layouts already exist in ETX since 2.5... And this is off topic,

raphaelcoeffic, 2022-07-06:
The main issue regarding widget options is that for now they use a fixed size array in memory, thus increasing the memory consumption for all models exponentially. So we would need to switch to a list with dynamic allocation instead, which is kind of non-trivial right now. But we should, indeed.

raphaelcoeffic, 2022-07-06:
Displaying these options is not the issue, we shall find a way ;-)

philmoz, 2023-04-22:
> The main issue regarding widget options is that for now they use a fixed size array in memory, thus increasing the memory consumption for all models exponentially.

Is memory really a problem, as this change only applies to color radios?

I tried increasing the widget option count to 10 on my TX16S.
RAM usage increased to ~78K.

AFAIK all color radios have 192K RAM so this is still using less than 1/2 what is available.

Dynamic memory (malloc/new) comes out of the heap which is in SDRAM.

PR #5365 (2024-08-05): feat(color): allow up to 10 widget options.
Fixes #1840 
Prerequisite for #1095