EdgeTX/edgetx #1844: Change the way backlight works on a slider/potentiometer to turn screen off 100% manually

EmrysTheWzrd, 2022-04-17:
I would like to see the backlight range increased back to the range it was before the 2.7 update when being controlled by a knob on the special functions page.

In 2.7, when the backlight is set on a slider/potentiometer there is no way to set the knob minimum value to equal backlight 100% off. You can only dim the radio screen but there is no way to turn it off completely. Before 2.7 I was able to use a slider to turn the radio screen all the way off while in use to save battery.

Proposed solution:

In the radio setup menu, there is already a backlight max and min value. The min value appears to be locked out when you don't have the backlight set to time out. I would like to see the min value be always adjustable and affect the backlight min value globally. The backlight can be set ON by default. Now, when you set the backlight to a knob in the special functions tab, the backlight will dim to the min value you set wether or not the backlight timeout is enabled in the radio setup menu.

pfeerick, 2022-04-18:
That is because the OFF value is what the backlight drops to when it times out, it's not the minimum value. And ON is what the current value is, not max. Maybe we need some more settings, for what the min and max values can be. 

Discussion in #1637 needs to be considered as well.

MervDale, 2022-04-18:
This issue with the screen brightness and battery worry seems to be ongoing, Can we find out by test or data just how much power these screens use, obviously it will vary between brands but if it turns out to be insignificant then it may sort out this issue once and for all. If this is the case then turning it down to minimum may satisfy those who don't like to look a a bright screen when flying ;-Q

pfeerick, 2022-04-18:
While I agree with the sentiment, I *do* think we have to fix the backlight situation properly... off should be an option still. 

But, for some rough data, based on the measurements from my Riden RD6018 PSU @8v (so by no means treat these as highly precise measurements - only rough real-world figures) driving a TX16 , with the backlight on full, overall power consumption drops from around 0.24A to 0.11A when the screen turns off. This is without any RF enabled. 

Turn on the internal MPM, add 0.07A to both of those figures. Use ES24TX ELRS with module set to 10mw, add 0.16A, and for 250mW, add 0.25A.

Power consumption of the display isn't insignificant, but you can see in the grand scheme of things the radio itself and the module will suck up most of the power. Also, it helps to know how LEDs work - the harder/brighter you drive them, the more power they suck up as the efficiency goes down. i.e. as a contrast, there was only a 0.01A difference in power draw between minimum brightness and backlight off (which you can still get with the auto off). 

For my own personal needs, 50% brightness is more than sufficient most of the time, and I only ramp it up to 100% outdoors in sunlight occasionally. If it only dropped to minimum brightness instead of OFF, I would be fine with that, as it would actually still be viewable when indoors. However, I still do have it set to auto off on both stick and keys input (depending on primary use of radio, others are set to keys only) as this also disables the touchscreen when the backlight is off (keys does also, as the first touch is consumed in waking the display up).

MervDale, 2022-04-18:
Another thing worth a mention is that the screens that will draw the most out of the onboard battery will be the Hi-Res colour screens and in most cases, if not all cases,  the Tx's with these screens will have removeable batteries so carrying a spare Rx battery would alleviate any worry of no useable Rx on your flying dayz  ;-Q

pfeerick, 2022-04-19:
Well, this issue actually only applies to colorlcd anyway... as B&W screens are allowed to fully turn off, as they don't *require* a backllight to be viewable ;)

JimB40, 2022-04-24:
someone posted that mosquitos are biting screen when it's on during night fly. ;)

vittorio0000, 2022-08-29:
Hi, is there any news for this feature? For me it is very important to turn off the screen completely

MervDale, 2022-08-29:
So far it is still an option to turn your off brightness to 0 in your backlight settings and your on brightness to what ever you like and your timer to the min of 5 sec and when that timer kicks in at 5 sec your screen will be completely black and only a touch of the screen and or KEYS (when set this way) will bring it back NOT the flight controls .  The issue was that you could turn your timer to 0 then 0 time with 0 brightness meant 0 screen and a pain to get it back. Hence the current settings .

vittorio0000, 2022-08-29:
> So far it is still an option to turn your off brightness to 0 in your backlight settings and your on brightness to what ever you like and your timer to the min of 5 sec and when that timer kicks in [...]

Honestly I would like to be able to adjust the brightness of the screen with a potentiometer from 0 to 100, it was possible to do it in the previous version and now not anymore, is it possible to reimplement this function?

MervDale, 2022-08-29:
So I guess being able to turn the screen  off indirectly doesn't suit your purpose?

raphaelcoeffic, 2022-08-30:
> Honestly I would like to be able to adjust the brightness of the screen with a potentiometer from 0 to 100, it was possible to do it in the previous version and now not anymore, is it possible to reimplement this function?

The change planned is to remove the "Special Function" for brightness, and be able to assign a knob directly. Once this is done, the knob regulate from OFF to ON brightness, regardless of any other setting. However, nobody had time to implement this for 2.8, so I'm afraid this will have to wait a bit.

vittorio0000, 2022-08-30:
> The change planned is to remove the "Special Function" for brightness, and be able to assign a knob directly. Once this is done, the knob regulate from OFF to ON brightness, regardless of any other [...]

Thanks for the replies.  Pity, it was a very convenient function in my opinion, I hope it does not fall into oblivion

raphaelcoeffic, 2022-08-30:
> Pity, it was a very convenient function in my opinion, I hope it does not fall into oblivion

The question is: which function does a SF give you, that the new config item won't? Can we live with this? Isn't it better to have some simple setting instead of something that needs a 20 minutes YT video so people can configure it?

raphaelcoeffic, 2022-08-30:
Using global functions / special functions is way above what the average user is able to do. This is why we want to change it into a simple setting, which, as I understand, would be enough for your use case as well.

vittorio0000, 2022-08-30:
> L'uso di funzioni globali/funzioni speciali è molto al di sopra di ciò che l'utente medio è in grado di fare. Questo è il motivo per cui vogliamo cambiarlo in un'impostazione semplice, che, a quanto ho capito, sarebbe sufficiente anche per il tuo caso d'uso.

I understand, I believe that at this point I will stick to 2.5. Thanks again

pfeerick, 2022-08-30:
Just so this is crystal clear - the new config option will allow exactly what the old SF did - without the need for a SF. And since it will be directly linked to the configured control, there shouldn't be any possibility that a bug/condition will cause you to have no display *at all* (untill you plugged the radio into a computer, and either used Companion to fix it, or altered the `radio.yml` file). 

This is the reason it is not allowed to completely turn off via a control atm - it was felt that it would be better that the display be dim, rather than not turn on, and thus render the radio **unusable** in some cases.

commit db410eb2 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit 8798b51e (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ba0fb945 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53
commit ab3cb457 (2022-08-31): feature(backlight): obsolete SF and allow for assigning a knob

Fixes #1844
Fixes #1637
Fixes #418
Fixes #53