EdgeTX/edgetx #1845: Companion crashes when importing OpenTX models

HerpDerpula, 2022-04-17:
When importing models from a jumper t16 pro running opentx 2.3.7 into the edgetx companion 2.7.0, edgetx companion will sometimes crash. I found this issue when attempting to migrate the opentx models from the t16 pro into the tx16s running edgetx 2.7.0 by copying the models/radio folder (I did try deleting the yaml files). 

When edgetx companion 2.7.0 crashes, it throws no errors. It just disappears. OS is windows 10 x64. Please reference the attached models and radio folder that is a direct copy off the t16 pro running opentx. Some of these models are configured to use inav luatelemetry scripts. 
 
[T16Pro_OpenTX_2.3.7_models_radio.zip](https://github.com/EdgeTX/edgetx/files/8502323/T16Pro_OpenTX_2.3.7_models_radio.zip)

elecpower, 2022-04-18:
Unable to reproduce using Ubuntu 20.04. Compiled with debug and loaded Companion:
- 2.7 branch
-- renamed zip as otx
-- opened with T16 Pro profile
-- changed profile to TS16S profile
-- closed models file
-- opened otx with TS16S profile
-- closed models file
-- copied zip to sd card, deleted yml files and Read Models and Settings from Radio
-- saved converted files to etx and reopened
- 2.8 nightly
-- same tests

Bash shell debug output appears normal.
If there is a reading/conversion incompatibility issue Companion usually crashes before displaying the list of models.
You report it does not happen on every occasion which may point to an OS environment issue.

HerpDerpula, 2022-04-18:
I will run further tests if needed. I can also post a screen recording of the issue if that helps. It's possible that it's my OS environment, is there a debug log for the EdgeTx companion?

pfeerick, 2022-04-19:
Yes... either when you run it from the command line, or you can turn on the logging to file...

https://user-images.githubusercontent.com/5500713/163895509-280def6b-c2f6-491a-bfb9-6484716f68dd.mp4

HerpDerpula, 2022-04-19:
[CompanionDebug_22-04-18_20-33-32.log](https://github.com/EdgeTX/edgetx/files/8508854/CompanionDebug_22-04-18_20-33-32.log)
Pretty consistently able to reproduce it, seems to occur most often when EdgeTX companion is minimized and then brought back from the task bar. The debug log seems out of order. Don't see any errors that make sense relating to the crashing of companion. 

Also, be advised that I do have the radio profile set to tx16s when reading models from the T16 pro running opentx

elecpower, 2022-04-19:
In Ubuntu using 2.8 nightly minimised and maximised multiple times and cannot get it to crash.
Log is messy due to opening an otx which needs to be converted to yml and read back but nothing abnormal spotted.

HerpDerpula, 2022-04-19:
It may be the radio and not the model files. I unplugged the radio last night and left EdgeTX open and we're at ~24 hours now without a crash with the radio files still cached on the screen. Does this make any sense from the way EdgeTX communicates with the OS and radio?

elecpower, 2022-04-20:
After importing Companion works on a consolidated temp etx file so no need for radio to be connected.
In your Companion radio profile is the sd card path mapped to the pc or radio? If radio try changing to pc leave radio connected and see if this helps.

elecpower, 2022-05-11:
@HerpDerpula is this still an issue or can we close?

elecpower, 2022-11-16:
Closed as no update from originator and no other reports

closed 2022-11-16 by elecpower