EdgeTX/edgetx #1852: No support for internal gyro on horus x12s

Luznatural, 2022-04-18:
We have been wayting for this issue for three years in opentx. All Horus users (Frsky X12S) are very upset with opentx. They abandoned us forever.
It would be nice to get internal gyro working on Frsky X12S using Egde TX
I kindly request to try it
Thanks in advance
V/R

gagarinlg, 2022-04-18:
I am not sure if any Dev has one, I do not have one.
What do you expect from the gyro? Just have the gyro axis as additional inputs, like sticks and sliders?

rotorman, 2022-04-18:
@Luznatural Work is underway in PR #1744 , please have just a little bit more patience.

Luznatural, 2022-04-19:
In theory, and always as advertised by frsky, this feature should permit to operate servos (i.e. gimbals for a camera) just by moving or inclining the transmitter. In other words, use the transmitter itself as a joystick.

1Sven, 2022-04-19:
The gyro means for me "Bierschalter"... pafleraf will do that in the near future, really, it is coming!

Luznatural, 2022-04-22:
Thanks to all of you on behalf of all Horus X12S users!!!

Luznatural, 2022-05-04:
Frsky X12S has an internal GPS and a 6-axis sensor (3-axis gyro and 3-axis accelerator) according the instruction manual and the advertised especificatrions.

closed 2022-06-25 by raphaelcoeffic

1Sven, 2022-06-25:
Oh, ein Traum geht in Erfüllung... Super!

Luznatural, 2022-06-27:
Thank you very much!!!