EdgeTX/edgetx #1854: 9xd+ model name corrupt after initial eeprom conversion

jj-uk, 2022-04-18:
After loading ETX for the 1st time, the conversion from OTX 2.3.14 completed and the radio started up.

Some, not all, model names are corrupted. 
See the images. This shows the OTX and the ETX models.

**To Reproduce**
Fresh install ETX over the top of OTX.

**Expected behavior**
Model names to be preserved.

**Screenshots**
[image]

**General information**
Taken from version screen in radio:
 - FW: edgetx-9xd+
 - VERS: 2.7.0 (f79978b6)
 - DATE: 2022-04-16 03:41:02
 - CFGV: 221
 
- Hardware Frsky X9D+ 2015

elecpower, 2022-04-18:
@jj-uk did you convert on radio or using Companion?
I loaded the zip into Ubuntu 20.04 builds of Companion 2.7 and 2.8 nightly and no corruptions

jj-uk, 2022-04-18:
@elecpower I converted on the radio when prompted.

elecpower, 2022-04-18:
@jj-uk thanks for the additional info

driedeker, 2022-04-19:
Does the same on my X9D+se2019 includes some curuption in mixes

driedeker, 2022-04-19:
Does the same on my X9D+se2019 includes some curuption in mixes etc

[image]
[image]
[image]

alm-nl, 2025-06-21:
Confirmed on X9D Plus, coming from OpenTX 2.3.15, installing EdgeTX 2.8.5 and doing the conversion on the radio. I also installed 2.9.x, 2.10.x and 2.11.x where x is the current latest of each version, although I don't think that matters because 2.8.5 did the conversion. Did not use EdgeTX Companion yet.
It seems to happen with model names that are longer than 10 characters.

The active model also had this in the Outputs menu, I have checked other models and they don't seem to have this behavior.

jj-uk, 2025-06-21:
I think EdgeTx dropped support for the x9d from v2.4, although it seems to list the same radio twice as supported and not suported?

https://edgetx.org/supportedradios/#not-supported-by-the-edgetx-team-community-supported-only

OpenTx probably does almost everything you'd need on the x9d anyway.

pfeerick, 2025-06-21:
No, it is supported as of 2.4, with 2.11 being the last version to support
that radio. However, it is only community supported since we have no
support from the manufacturer.