EdgeTX/edgetx #1855: 9XD switch position text missing

jj-uk, 2022-04-18:
The switch designators, e.g. SA !SA, SB, etc including the stick names are missing throughout the firmware.

See images below.

**To Reproduce**

**Expected behavior**
Switch and stick names should be displayed.

**Screenshots**

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**General information**

Taken from version screen in radio:
 - FW: edgetx-9xd+
 - VERS: 2.7.0 (f79978b6)
 - DATE: 2022-04-16 03:41:02
 - CFGV: 221
 
- Hardware Frsky X9D+ 2015

pfeerick, 2022-04-18:
Can you also upload at least your `RADIO\radio.yml` - better still if you can do both MODELS and RADIO folders... as I suspect there is something wrong in the formatting of the config file as I'm not seeing this on the X9D+2019 - which as far as the UI code is concerned should be exactly the same.

jj-uk, 2022-04-18:
No problem. Find attached. 
I tried to change the splash screen timeout in companion, then write the models back to the radio. That didn't change the timeout value in the radio. It seems that the splash screen timeout isn't saved in the radio yaml file. Not sure if me doing that broke the file?

[Models and Radio.zip](https://github.com/EdgeTX/edgetx/files/8508655/Models.and.Radio.zip)

pfeerick, 2022-04-19:
Ok, so the reason your switch names are "missing" is because they have been defined as blank - three SPACES:
[image]

Same for the sticks, pots and sliders... 

To fix that, simply remove the spaces between the "" - and radio will use the default rather than "custom" name. I've just done a find and replace if you want to drop this radio.yml back on the radio (after unzipping)...
[radio.zip](https://github.com/EdgeTX/edgetx/files/8508734/radio.zip)

For power delay (pwrOnSpeed, pwrOffSpeed), they are both 0, which means two seconds (default 2 seconds - value chosen) ... you can change that to `-1` if you want the maximum 3 second delay, or 1 or 2 to reduce the power delay to 1 or 0 seconds respectively... if the radio isn't correctly updating them, that is.

jj-uk, 2022-04-19:
Thanks. I didn't make those changes to the text, so there must be a bug somewhere that's caused it. I'll try to reproduce the problem and report back.

closed 2023-05-26 by gagarinlg