EdgeTX/edgetx #1856: X9D+ (2015):  splash screen and backlight color incorrect on startup

malinhiles, 2022-04-19:
**Describe the bug**
-On radio startup, the "EdgeTX" splash screen is shown for 2 seconds even though it is disabled in the radio settings. 
-While the radio is displaying this splash screen, the screen backlight is (incorrectly) the stock "deep blue" hue, and not the "white" hue as set by user in radio settings. After the radio has booted past the splash screen, the user defined "white" backlight color resumes.

**Steps to reproduce the behavior:**
1. Turn on radio
2. Observe startup behavior

**Expected behavior**
On radio startup, the "EdgeTX" splash screen should not be shown if disabled in radio settings. If the splash screen were to be enabled by the user, the radio should use the user defined backlight color from radio settings. 

**Screenshots**
 Here are some quick photos of the issue. 
[image]
[image]

**General information**
 - EdgeTX version 2.7.0, commit 7ce8738
 - FrSky X9D+ 2015

pfeerick, 2024-01-08:
Is this still an issue? There have been quite a few changes around startup and splash screen code, and I've never actually experienced this myself on X9D+ (I don't think).

malinhiles, 2024-01-10:
Can no longer confirm on my own radio, I sold it 9 months ago.

closed 2024-01-10 by pfeerick