EdgeTX/edgetx #1857: X9D+ (2015) can't enter bootloader mode

malinhiles, 2022-04-19:
**Describe the bug**
By holding the mode 2 "yaw" and "aileron" trim switches together and turning on the radio, the radio should attempt to boot into bootloader mode. Instead, when holding those trim switches together, my radio attempts a normal boot to the radio home screen and fails with a "Key Stuck" warning. I experienced this bug for this radio with edgetx v2.5.0 and v2.7.0, but did not find this issue in v2.6.0.

**To Reproduce**
1. Hold mode 2 "yaw" and "aileron" trim switches together. 
2. Turn on the radio. 
3. Wait for bootloader screen while continuing to hold the trim switches.
4. Radio proceeds to attempt a normal boot but fails due to "key stuck" warning

**Expected behavior**
Radio would boot into bootloader mode when booted holding corresponding mode 2 "yaw" and "aileron" trim switches together.

**General information**
 - EdgeTX version 2.7.0 commit 7ce8738
 - Hardware Frsky X9D+ 2015

spiky1, 2022-04-25:
I have a 2015 X9D+ with EdgeTX version 2.7.0 (f79978b6), the bootloader starts fine, no problems.

pfeerick, 2022-04-25:
Have you verified that moving the trims inwards is functioning when using the radio normally? i.e. It's not a sticky or dodgy trim? I also have no problem with entering bootloader on X9D+2019

malinhiles, 2022-04-26:
> Have you verified that moving the trims inwards is functioning when using the radio normally? i.e. It's not a sticky or dodgy trim? I also have no problem with entering bootloader on X9D+2019 with v2.7.0 and matching bootloader.

Yes, I verified that my trims function as expected when using the radio normally. I've been finding the issue to be consistent, and reliably present when I flash edgetx v2.5.0 and v2.7.0. Across multiple attempts, I couldn't successfully get into bootloader mode. In edgetx 2.6.0, I can get into bootloader just fine.

Just to make sure, I know you mentioned you9are on the 2019 taranis. I am on the 2015 acccst taranis plus, not the 2019.

pfeerick, 2022-04-26:
Yes, I understand that, but the prior commenter was also on x9d+, and had no problem. I also have a x9d+, but am waiting for a new mainboard for it, so will be a couple weeks before I can test that. 

So the next thing to check is when you flash edgetx, are you updating the bootloader? i.e. do you flash using DFU (Flasher, Buddy, etc) or do you update via the bootloader (and also update the bootloader from within the firmware).

malinhiles, 2022-04-26:
> Yes, I understand that, but the prior commenter was also on x9d+, and had no problem. I also have a x9d+, but am waiting for a new mainboard for it, so will be a couple weeks before I can test that.

Oh ok, no worries, just wanted to check.  

> So the next thing to check is when you flash edgetx, are you updating the bootloader? i.e. do you flash using DFU (Flasher, Buddy, etc) or do you update via the bootloader (and also update the bootloader from within the firmware).

Yea, I'm flashing and updating the bootloader simultaneously via the companion on each flash. I had to use this method when moving from 2.5.0 to 2.6.0 and from 2.7.0 back to 2.6.0 because I couldn't get access to the radio's bootloader; i'm currently back on 2.6.   In initially moving from 2.6.0 to 2.7.0,  I used the same update method (simultaneous firmware and bootloader update via companion) for sake of consistency. 

Long story short, for each edgetx version that I flashed (2.5.0, 2.6.0, 2.7.0), I simultaneously updated to the corresponding/matching bootloader(2.5.0, 2.6.0, 2.7.0).

malinhiles, 2022-05-27:
Is there a solution to this?

gagarinlg, 2023-05-05:
can anyone confirm that this is still an issue?

malinhiles, 2023-05-10:
Can confirm this issue exists in edgetx 2.5 and 2.7. Haven't checked in the more recent edgetx releases

pfeerick, 2024-07-06:
I think we will have to chalk this up to being some sort of hardware issue, as I have two X9D+ now, and neither have ever exhibited this issue (don't ask) - one has been running pretty much every release from 2.4 onwards, and the other from 2.7 or 2.8 onwards. 

In the SYS/radio -> Hardware -> Keys debug screen, are all the trims and buttons work properly? As the stuck key warning is suggestive that something is wrong, perhaps with one of the trims. It may be worth opening the TX up, and reseating the ribbon cable, as I have had a single switch stop responding on the X9D+2019 due to it being knocked of a desk - and the X9D+ also uses ribbon cables to connect the mainboard and the RF, switch and trims board.

malinhiles, 2024-07-15:
Copy that. I appreciate the help and suggestions. I no longer have the X9D radio that I referred to when I made previous posts in the thread. I've moved over to a more frequently supported and popular radiomaster radio.

closed 2024-07-15 by pfeerick