EdgeTX/edgetx #1858: X9D+ (2015) alphanumeric characters in inconvenient order

malinhiles, 2022-04-19:
**Describe the bug**
This may be a "bug", but I'm not sure. Nonetheless it's a little bit clunky.

When editing model parameters that use alphanumeric symbols, like "Model Name," "Input Name," "Line Name," "Mix Name," etc., the user uses the "+" key to cycle through a list of symbols, numbers, and letters for a selected character slot. This list seems to be in an inconvenient order for creating model names, line names, and mix names in comparison to the list order used in OpenTX.

In EdgeTx on my radio, the user has to click through all of the symbols and numbers just to get to the letters: 
" _ - 0 1 2 3 4 5 6 7 8 9 0 A B C D E F G H I J K L M N O P Q R S T U V W X Y Z a b c d e f g h i j k l m n o p q r s t u v w x y z"

And because the capital letters are available sooner in the list than the lowercase letters,  it's easier but somewhat frustrating to use all capital letters and have to hold the "enter" key to make all but the first letter of each word lowercase after the fact, which is pretty clunky, all things considered. 

In OpenTx on my other radios, the user has access to the lowercase letters right at the top of the list, and can the user can hold "enter" to capitalize them if needed:
" a b c d e f g h i j k l m n o p q r s t u v w x y z 0 1 2 3 4 5 6 7 8 9 0 _ - , . "

Also, the EdgeTx radio character list is missing the comma and period symbols

**Expected Behavior**
It would be nice if the alphanumeric character list was in an order that made more sense and was more usable. On the X9D+ 2015 (what I mainly use) it's especially noticeable because the radio doesn't have a scroll wheel to zip through the list quickly; you have to nearly click through each character with the "+" and "-" buttons.

**General information**
 - EdgeTX version 2.7.0 
 - Hardware Frsky X9D+ 2015

dougshell, 2022-06-01:
As a new user (first-time pilot) I found this to be incredibly frustrating. The scroll sensitivity also seemed to be far too strong as well (commonly scrolling past target letter) but this may be outside the scope of this "issue"

pfeerick, 2023-12-06:
@3djc What determines the order of the characters for B&W?

philmoz, 2023-12-06:
> @3djc What determines the order of the characters for B&W?

Alphabetical from ' ' to 'z'.

radio/src/gui/common/stdlcd/draw_functions.cpp - editName and isNameCharset functions.

pfeerick, 2023-12-06:
Thanks. So that would mean it's due to #12. 

This is the equivalent block in OTX... I'm thinking something was altered wrong when removing ZCHARS... but cant place a finger on what atm. It's easy enough to add the missing symbols via `isNameCharset`, and could inhibit either uppercase or lowercase except via toggle by making one or the other return false, but I wonder if instead we should be controlling the order of the list, rather than just blindly in ASCII numerical order...

https://github.com/opentx/opentx/blob/a26c80a695725419078cc73fedeb9d4728bf2fa5/radio/src/gui/common/stdlcd/draw_functions.cpp#L123-L241

3djc, 2023-12-06:
I thought all this time it was a change made on purpose !

pfeerick, 2023-12-06:
Me too, but it also didn't make much sense ... I would have at least preferred the numbers after the lowercase letters if it were intensional, given I already knew you could toggle uppercase (or uppercase and toggle lowercase... IMO makes very little difference which around it goes as people will argue they use all uppercase, etc.). It was one of those "will look into it someday" things so I was glad when someone opened this issue making it more likely it would be looked into.

raphaelcoeffic, 2023-12-07:
> I thought all this time it was a change made on purpose !

Nope, nothing intentional here 😅

PR #5390 (2024-08-08): fix(bw): change char selection order when editing names
Fixes #1858 and fixes #4618 

Change the order of character selection to:
```C
        " abcdefghijklmnopqrstuvwxyz0123456789-,."
```

Use long press to change case.
closed 2024-08-08 by pfeerick