EdgeTX/edgetx #1865: TBS Agent Lite on x9d/x9d+ causing Storage Warning EEprom Data V220 / Bad radio data

spyder0069, 2022-04-20:
EdgeTX version 2.7.0
x9d and x9d+
tbs crossfire full size with oled firmware 6.17

When using the tbs agent lite (v0.96) and going into XF Transmitter then radio settings and toggling options after a random number of toggling the radio will give a speeker click and show Storage warning eeprom data v220.  Then if you hit enter it will go to a Bad radio data message.  Then it will repeat this loop.  You have to cycle the power to correct it.  I have both a x9d and x9d+ and it happens with both radios.

JimB40, 2022-04-24:
Thx for posting. Observered this too. Seems like something is wrong with memory management for LUA as X9D pretty stable 60kB for LUA. @pfeerick you have x9d?

apeelme, 2022-06-23:
> EdgeTX version 2.7.0 x9d and x9d+ tbs crossfire full size with oled firmware 6.17
>
> When using the tbs agent lite (v0.96) and going into XF Transmitter then radio settings and toggling options after [...]

same to me...but TBS Tracer
EdgeTX version 2.7.0
x9d+ ACCST

borisbl4, 2022-08-03:
I have the same problem, I understand there are no solutions yet?

pfeerick, 2022-08-05:
I've got X9D+ and X9D+2019. With TBS Agent 0.96 and Tracer module, it seems related to what menu I go into. i.e. Agent Lite loads fine, can go into the device with no problem. Go into the about screen and leave it there for a while, not problem. team racing, view and change a setting there. seems fine. video tx, view and edit settings seems fine. MAVLink, view and change setting seems fine. It's when I'm in radio settings (or have been in radio settings) it seems more to prone to crashing. However, if I use the Crossfire Lua, it will crash pretty quickly also after loading device settings. On the X9D+ it is like it has run out of memory... i.e. I opened and close TBS agent lite, and only have 2100b free mem (according to statistics debug menu)... but I would have thought the lua would crash/halt, not the radio. 😮

Hellfire82, 2023-09-06:
I've just upgraded to EdgeTX 2.9 & have encountered this issue. Obviously looking at the above comments, it's not a new issue relating to the new firmware. However I don't recall encountering this issue before & I've been using EdgeTX since V2.5.
Like @pfeerick stated, the 'storage warning' only seems to happen when scrolling through the 'radio settings' sub-menu of the Micro TX 🤷‍♂️

raphaelcoeffic, 2023-09-07:
Are we deal with a file system corruption? Or is that in memory, and propagating to the radio file?

Hellfire82, 2023-09-07:
> Are we deal with a file system corruption? Or is that in memory, and propagating to the radio file?

I thought that the cheap SD card that came with my radio might have been the culprit, so I replaced it with a Sandisk 32GB SDHC U1.
I fully formatted it then copied across fresh versions of the v2.9 SD card contents, BF LUA & TBS Agent Lite 0.96, as well as re-flashing the bootloader & firmware to v2.9.

So I don't think any of my files were corrupted since the issue occurred on multiple SD cards.

raphaelcoeffic, 2023-09-11:
> So I don't think any of my files were corrupted since the issue occurred on multiple SD cards.

I'm more thinking of some LUA function accessing/overwriting memory regions it shouldn't and thus corrupting the internal RAM which finally might trash some settings persisted to the SD card.

When this happens, it would be interesting to know what is actually on the SD card. So when the message with "Bad radio data" comes, don't go further, switch the radio OFF instead. Then inspect the SD card with your computer. We would need to see the file located at `/RADIO/radio.yml`, as this is what the software is trying to read to get your configuration.

Hellfire82, 2023-09-11:
@raphaelcoeffic, as requested, when the "Bad radio data" screen appeared, I switched off the radio & copied the radio.yml file from my SD card. It is available to download via the following link:

https://drive.google.com/file/d/1IawldFpmw1Yn7RSHUYPEU3CBd5wjF10f/view?usp=drivesdk

Let me know if I can be of any more help.

raphaelcoeffic, 2023-09-11:
The payload looks ok to me and loads fine in the simulator.

For reference:
```yaml
checksum: 27632
manuallyEdited: 0
semver: 2.9.0
board: x9d+
calib: 
   Rud:
      mid: 903
      spanNeg: 670
      spanPos: 723
   Ele:
      mid: 879
      spanNeg: 490
      spanPos: 775
   Thr:
      mid: 1058
      spanNeg: 755
      spanPos: 703
   Ail:
      mid: 1004
      spanNeg: 825
      spanPos: 808
   POT1:
      mid: 1057
      spanNeg: 1041
      spanPos: 975
   POT2:
      mid: 1058
      spanNeg: 1042
      spanPos: 974
   SLIDER1:
      mid: 1032
      spanNeg: 899
      spanPos: 806
   SLIDER2:
      mid: 1009
      spanNeg: 994
      spanPos: 812
currModel: 0
contrast: 0
vBatWarn: 65
txVoltageCalibration: 0
backlightMode: backlight_mode_on
antennaMode: MODE_PER_MODEL
disableRtcWarning: 0
keysBacklight: 0
internalModule: TYPE_XJT_PXX1
trainer: 
   mix: 
      0:
         srcChn: 2
         mode: REPL
         studWeight: 100
      1:
         srcChn: 3
         mode: REPL
         studWeight: 100
      2:
         srcChn: 1
         mode: REPL
         studWeight: 100
      3:
         srcChn: 0
         mode: REPL
         studWeight: 100
view: 0
fai: 0
beepMode: mode_nokeys
alarmsFlash: 0
disableMemoryWarning: 1
disableAlarmWarning: 0
stickMode: 1
timezone: 0
adjustRTC: 0
inactivityTimer: 0
internalModuleBaudrate: 0
splashMode: 2
hapticMode: mode_nokeys
switchesDelay: 0
lightAutoOff: 2
templateSetup: 17
PPM_Multiplier: 0
hapticLength: 2
beepLength: 2
hapticStrength: 2
gpsFormat: 1
speakerPitch: 0
speakerVolume: 23
vBatMin: 60
vBatMax: 88
backlightBright: 0
globalTimer: 1143617
bluetoothBaudrate: 0
bluetoothMode: OFF
countryCode: 2
pwrOnSpeed: 0
pwrOffSpeed: 0
noJitterFilter: 1
imperial: 0
disableRssiPoweroffAlarm: 1
USBMode: 0
jackMode: 0
ttsLanguage: "en"
beepVolume: 4
wavVolume: 4
varioVolume: 4
backgroundVolume: 4
varioPitch: 0
varioRange: 0
varioRepeat: 0
customFn: 
   0:
      swtch: "ON"
      func: PLAY_SCRIPT
      def: "bfbkgd,On"
   1:
      swtch: "ON"
      func: VOLUME
      def: "POT2,1"
slidersConfig: 
   SLIDER1:
      type: with_detent
      name: ""
   SLIDER2:
      type: with_detent
      name: ""
potsConfig: 
   POT1:
      type: with_detent
      name: ""
   POT2:
      type: with_detent
      name: ""
backlightColor: 20
switchConfig: 
   SA:
      type: 3pos
      name: ""
   SB:
      type: 3pos
      name: ""
   SC:
      type: 3pos
      name: ""
   SD:
      type: 3pos
      name: ""
   SE:
      type: 3pos
      name: ""
   SF:
      type: 2pos
      name: ""
   SG:
      type: 3pos
      name: ""
   SH:
      type: toggle
      name: ""
bluetoothName: ""
ownerRegistrationID: ""
rotEncMode: 0
uartSampleMode: 1
radioGFDisabled: 0
radioTrainerDisabled: 1
modelHeliDisabled: 1
modelFMDisabled: 1
modelCurvesDisabled: 1
modelGVDisabled: 0
modelLSDisabled: 0
modelSFDisabled: 0
modelCustomScriptsDisabled: 1
modelTelemetryDisabled: 0
```

raphaelcoeffic, 2023-09-11:
> it seems more to prone to crashing

@pfeerick so what is happening is actually some kind of memory access violation, which triggers a crash + reboot, and in that context, the SD card is not accessible, so that the system believes it should read/convert from EEPROM? Or is that not really a reboot? Like after you go on with the "Bad radio data" screen, does it have the `!` in the corner? Or maybe it's just triggering the same kind of file system reload like after USB MSD disconnection?

jporter100, 2024-06-16:
Not sure if this information is of use but I only see with the "tbs crossfire full size with oled" same as the OP. I also have a Crossfire MicroTX II and it never encounters the problem using the same radio. With the full size Crossfire module it happens everytime I've been into an affected submenu.

Taranis X9D+ SE (the ACCST version)
EdgeTX info shows as:
FW: edgetx-x9d+
VERS: v2.8.4
NAME: FlyingDutchman
Crossfire Full - v6.19
Crossfire MicroTX II - v6.19

Thanks