EdgeTX/edgetx #1868: Cannot quit the “control warning” popup for S1 and S2.

Daannoa, 2022-04-20:
When the radio start, for switches, move the switch in the good position will quit/remove the “control warning” popup But for the pots (S1 and S2), moving it in the right position don’t remove the popup.
(However I can skip it with the arrow in the back right corner)
I think the behavior should be the same for pots as for switches but maybe I'm wrong…

(This is maybe link to this issues:
https://github.com/EdgeTX/edgetx/issues/1843 )

**General information**
 - EdgeTX version 2.7.0
 - Hardware TX16S

eshifri, 2022-04-21:
I cannot reproduce it.

Daannoa, 2022-04-21:
After flashing the firmware again and after redoing the calibration (twice), it seems to work.

So the problem seems to come from me, sorry

closed 2022-04-22 by Daannoa