EdgeTX/edgetx #1871: [Bug]: Companion internal module cant be changed (Companion 2.7)

jasc76, 2022-04-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Current Behavior

[image]
I try to set multi in radio profile but it always reverts back to CSRF. 
2.7 Release Version
RM Zorro 4-1 

### Expected Behavior

Set internal module to MULTI 

### Steps To Reproduce

Open radio profile
set internal module
leave
open profile again
module is set back to CRSF

### Version

2.7.0

### Transmitter

Radiomaster Zorro

pfeerick, 2022-04-21:
I can reproduce this, and is not limited to the Zorro, also applies if for instance you try to set TX16S to CRSF. Upon re-openeing the profile page it will revert to MULTI. 

[image]

jasc76, 2022-04-21:
I also tried by loading settings from radio and hitting "save to current profile". Also does not work

PR #1873 (2022-04-21): Fix companion radio profile default module resetting to board default
Fixes #1871
closed 2022-04-21 by pfeerick