EdgeTX/edgetx #1874: Frsky X9lite doesn't see the GPS

kusygacek, 2022-04-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a Frsky X9 Lite radio, to which I uploaded the EdgeTX Black Pearl software. Previously, I had OpenTX installed and everything was fine. Now, after installing EdgeTX, I lost my GPS sensors. The GPS is connected to the Fettec Kiss G4 controller. Receiver is TBS sixty9 , no SmartAudio.

### Expected Behavior

Get back GPS sensors

### Steps To Reproduce

I did the steps but don't see any error.

### Version

2.7.0

### Transmitter

FrSky X9 Lite / Lite S

pfeerick, 2022-04-22:
What happens if you run sensor discovery on the telemetry page... does it pick the GPS up then? If so, maybe there is a glitch in the conversion regarding the ID. What firmware on the FC? As I know betaflight gps telem is still working.

kusygacek, 2022-04-22:
I have removed existing sensors many times and initiated a search for new ones, but the result is still the same, no GPS sensors. I only see 15 standard sensors. The firmware on the FC is the newest version - FETtec FC G4 1.3-RC47m. 
As for Betaflight, you are right. I have a second drone, FC Matek F405STD, TBS crossfire nano receiver, Matek GPS type SAM M8Q and Betaflight 4.3. Everything works fine. Only the FM sensor (Flight Mode ?) shows an error - in the case of FC Kiss, there is no FM error (Flight Mode = level).