EdgeTX/edgetx #1875: ETX 2.7/Zorro - Logical switch corrupted after transfer to Zorro

RC-SOAR, 2022-04-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

A logical switch created in Companion is incorrect after upload to the Zorro.

### Expected Behavior

Expect to see the same definition of logical switch on the actual transmitter.

### Steps To Reproduce

1. Create a model in Companion 2.7 with Zorro profile, with one logical switch:
  L01 **a<x, V1=Ele, V2=-90**
2. "Read models and settings from radio"
3. Drag new model to Zorro window
4. "Write models and settings to radio"
5. Now start the tx, and examine L01. V1 should be **Ele**, but is incorrectly translated as **[I]Ele**

### Version

2.7.0

### Transmitter

Radiomaster Zorro

pfeerick, 2022-04-22:
Can you double check I'm not doing something significantly different here? As I'm unable to reproduce this atm... mine seems to be correctly using the Stick Ele vs Inputs Ele

https://user-images.githubusercontent.com/5500713/164679777-57f401f0-73ac-4014-9c02-6f898e5d014d.mp4

[image]

RC-SOAR, 2022-04-22:
Apologies for any wasted time, it was my mistake! 

A combination of wrong stick mode, and incorrect interpretation of the stick symbol... doh! 

I shall close the issue now.

closed 2022-04-22 by RC-SOAR

pfeerick, 2022-04-23:
No problem Mike, it happens! :)