EdgeTX/edgetx #1876: new Gauge lua widget

offer-shmuely, 2022-04-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have always frustrated from the poor quality existing gauge  

### Describe the solution you'd like

I am working for some time on a new widget
it's new Gauge, with a needle, like to old days analog gauge

it have automatic min/max indication
it now have (a simple) app-mode as well
I added preset for the min/max values for predefined known telemetry

it is not super stable yet, that's why I do not pull-request it yet
I like to get your feedback and suggestions

comments are welcome

you can see the code here: 
https://github.com/offer-shmuely/opentx-2.3-x10-widgets/tree/master/Gauge2

offer-shmuely, 2022-04-22:
[image]

offer-shmuely, 2022-04-22:
App Mode

[image]

[image]

offer-shmuely, 2022-04-22:
can admin assign it to me?

pfeerick, 2022-04-23:
Can only assign if someone is a member of the project. However, you can assume you it's yours ;)

First impression (from the screenshot only) - gorgeous! I especially like how it splits the dials out in fullscreen mode. and is able to show min/max/current value when as a single dial. I'll certainly play with the actual widget later and see how it goes.

JimB40, 2022-04-24:
@pfeerick  Don't we do "LUA widget" path? Like wirh FM2M widgets panck?

offer-shmuely, 2022-04-24:
@JimB40 can you please explain your comment?
What is “lua widget path”?

pfeerick, 2022-04-24:
I think he means the [lua-scripts repo](https://github.com/EdgeTX/lua-scripts), which we are using to try and consolidate a list of the various Lua scripts and widgets that are around, rather than people having to go search high and low for them. Later, that can feed it making it so they can be offered for easy install (but first we need to know where and what they are). 

I'm not sure yet exactly what would happen with this one... I don't think it is generally a good idea to distribute a Lua version of every built-in widget, but at the same time, if this significantly improves on the built-in one, then it would be good to make it readily accessible. I sort of think we need to have an extras pack with other maintained widgets, until Lua scripts and widgets can basically one-click installed via Flasher and/or Buddy. i.e. the sdcard pack is the minimal install, and then there is an extras pack you can install to get all the rest. If it were updated once a month, so overhead to maintain it wouldn't be that much.

JimB40, 2022-04-24:
Sorry thought you were trying to include it in C++. Thats okay.
Here is repo
https://github.com/EdgeTX/lua-scripts

pfeerick, 2022-04-24:
Yeah, Offer does Lua widgets :)

I've got no intentions on doing anything with the C++  widgets from at this time. There may be changes with LVGL, and I do want to increase the number of options widgets generally can have, but this is all.

wimalopaan, 2022-04-26:
Looks like the original link isn't correct.

pfeerick, 2022-04-26:
It was at the time... the widget has probably been split into two - GaugeDash3 and GaugeRotary
https://github.com/offer-shmuely/opentx-2.3-x10-widgets/tree/master

offer-shmuely, 2022-04-26:
Hi
not spited, just renamed to GaugeRotary

The GaugeDash3 is for testing capability of multiple gauge in a single dashboard, and check sizes (run it, it fun to watch)

I did a pull request so more user can test it:
https://github.com/EdgeTX/edgetx-sdcard/pull/60

offer-shmuely, 2022-05-15:
Merged to main branch
Waiting to happy users

closed 2022-05-15 by offer-shmuely