EdgeTX/edgetx #1877: Zorro with Ghost

FrippeF, 2022-04-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radiomaster Zorror elrs will not work with Irc Ghost module.

### Expected Behavior

The Radiomaster zorro elrs will take with Icr Ghost module

### Steps To Reproduce

It start the module. But nothing happend on the in stick on the module

### Version

2.7.0

### Transmitter

Radiomaster Zorro

FrippeF, 2022-04-23:
Don't have the module in tx auto mode.

closed 2022-04-23 by FrippeF

pfeerick, 2023-05-23:
> The external module is able to bind and communicate with the RC but there is no communication between the RC and IRC Ghost External Module.

Do you mean that you were able to bind (using the module's own controls) and the module can communicate with the RX, but there is nothing happening between the module and the handset/RC?

Have you tried what the prior user mentioned - go into the menu on the module, go into Mode, and change Tx Mode from Auto to Manual, and then set the protocol to GHST. I noticed on my JR Micro sized Ghost module I was having some issues when trying that just now, and it was somehow suggesting CRSF was a valid protocol to choose when I first change it from Auto (and it was non-responsie to the handset)?! Then it correctly only offered up GHST and SBUS (1.0.5.4 firmware).