EdgeTX/edgetx #1885: Create model Wizard missing Delta Wing/Flying wing option

CorsairPilot, 2022-04-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I use model wizard to create a new model it does not have a Delta wing/Flying wing option as Open TX did.

### Describe the solution you'd like

Add the Delta wing/Flying wing option into the wizard

### Describe alternatives you've considered

1. create model from scratch. Easy for some people but not for beginners.

### Additional context

I have none

pfeerick, 2022-04-23:
I'm going to close this, as it actually relates to the lua wizards, and this is already being tracked here... https://github.com/EdgeTX/edgetx-sdcard/issues/26 so will happen at some point. 

However, I am right in assuming this is for B&W though, as colorlcd didn't have a delta wizard (on radio). On Companion there are still new model wizards for both. And we still need to add the new helli stuff also. 😮

CorsairPilot, 2022-04-23:
> I'm going to close this, as it actually relates to the lua wizards, and this is already being tracked here... EdgeTX/edgetx-sdcard#26 so will happen at some point.

No this is on the color screen. I don’t know why that would make a difference though. I know how to set up a flying wing but someone I was helping is not that knowledgeable. I can from OTX and it had that in the wizard. Was surprised EdgeTX didn’t. 

Thanks

pfeerick, 2022-04-24:
It makes a difference because colorlcd and B&W have different capabilities, thus different options ;)

What version of OpenTX and SD card pack is this with... I know the X7/X9 B&W transmitters have a delta wizard (e.g. [X7 wizard luas](https://github.com/opentx/opentx/tree/2.3/radio/sdcard/taranis-x7/SCRIPTS/WIZARD)), but I don't believe the colorlcd ones ever did. AFAIK it was only on Companion as the second question of the Plane wizard - and this seems to be confirmed by the OTX guys themselves back in 2018 - https://github.com/opentx/opentx/issues/5713. 

Now, I know since then, the Multicopter and Heli wizards have been added, but still no Delta wing... but if you can point me in the direction of where it exists, I'll gladly add it, as it means I don't need to create it (and thus close that issue).

Eldenroot, 2022-04-24:
Would be really cool to add these into wizard... it could save a lot of time and questions for the beginners

CorsairPilot, 2022-04-24:
OTX 2.3.14 model setup wizard. I use glider no motor 2 ailerons channel 1 right, channel 2 left. No flaps. Tail V channel 1 right V, channel 2 left V. 
This gives the settings to run the flying wing configuration. 
EdgeTX using the same procedure gives a result that does not work. 
I’ll send pictures on another email as I can’t add them to this one. Or I don’t know how is a better statement. 

Phil

CorsairPilot, 2022-04-24:
Hope this helps

Phil

pfeerick, 2022-04-25:
Email attachments don't come through on github, so if you were trying to send something, you'll either need to zip it and attach it directly on github, or upload it to a site like https://filetransfer.io/ and share the link.

pfeerick, 2022-04-25:
If I use the glider wizard for (so there is no delta wizard on colorlcd OTX then?) both OTX (2.3.15) and ETX(2.7.0) give the exact same mixer config, and channel output is responding the same. 

[image]

However, it should be easy enough to split that functionality out of the glider wizar

CorsairPilot, 2022-04-25:
Your mixers are perfect. I’m on 2.5 ETX. Maybe that was the problem. I’ll upgrade to 2.7.
Thanks for what you guys do. OTX was getting stale after my 7 years of use, now I’m excited again. 

Phil

closed 2022-08-10 by pfeerick