EdgeTX/edgetx #1888: Scripts in telemetry mode do not receive PAGE key press events on RM Zorro

JimB40, 2022-04-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

PAGE keys are used for switching telemetry screens.
If one of the screen is script it is recieveng  EVT_VIRTUAL_NEXT_PAGE & EVT_VIRTUAL_PREV_PAGE but not on Zorro target.

If PAGE_NEXT or PAGE_PREV is used in script OS will ovverride action switching to the next telemetry screen but still user can have only one screen with script and then actions will not be ovverriden.

It works like that on other B&W targets.

### Expected Behavior

LUA telemetry scripts should  should receive EVT_VIRTUAL_NEXT_PAGE & EVT_VIRTUAL_PREV_PAGE events

### Steps To Reproduce

Run any script that catches key events and check for PAGE one

### Version

2.7.0

### Transmitter

Radiomaster Zorro

raphaelcoeffic, 2022-11-01:
This discussion is very old. The issue is the following: we have multiple telemetry pages. Now how do you want to make the difference between changing the telemetry page and the changing the page within the telemetry script? This is the reason why the `PAGE` key will stay with the system in that particular case, at least until the navigation between telemetry pages uses that key.

JimB40, 2022-11-02:
I know. It's kind of 'bug' in other targets. PAGE event is captured both OS and LUA. So if you have more than one telemetry screen it exits current script and does to another telemetry screen (script). Although dut to limited memory people usualy run only one telemetry screen with "big' script like BF Config, ToolBox or iNav.

I was thinking to bind PAGE_NEXT_LONG to switch telemetry pages while PAGE_NEXT is passed to LUA
Problem is with old FrSky targets where there is no PAGE_PREV and it is simulated with PAGE_NEXT_LONG.
Still better.

closed 2025-05-16 by JimB40