EdgeTX/edgetx #1892: Counter - announcement (sound/voice/vibrate)

Eldenroot, 2022-04-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I used Spektrum radios before I jumped to OpenTX and then to EdgeTX.

I would like to have an option to have this:
- beep every XX seconds (lets say on 30 seconds)
- announce (voice) time left every minute, for last 1 minute every 10 seconds 
- vibrate on every beep (point 1) and then vibrate for the every event  (point 2)

there are more options on Spektrum radios for these timers, this is only a basic example. Or is that possible (all) right now and I just do not know how? Thank you

### Describe the solution you'd like

Check above

### Describe alternatives you've considered

Lets discuss

pfeerick, 2022-04-25:
Some of this _is_ possible with logical switches and logical functions, but why should have have to do that when you can have nicer options that do this for you :)

1) beep every 30 seconds... should be possible with either 1 or 2 logical switches and a special function for the beep
2) voice prompt for every minute (enable the Minute call option), but for every 10 seconds in last minute no... only has the countdowns
3) not possible with current options... you can have voice *or* haptic, not both. 

Since the timer settings is going to being broken out to their own screens... thus the model setup page will not longer be three miles long for the timers (which, lets be honest, you set them up and then forget about them... so why do they need it be in the way all the time?)... at this point, we can consider adding more options. 

i.e. 'Vibrate on annouce' , more granular control over final countdown (every second, every 5 seconds, every 10),

Eldenroot, 2022-09-27:
For me just now as a basic improvement would be really nice to have an option to vibrate (haptic) + voice alert together... why we cannot use them both at once? More improvements are welcome :)

Eldenroot, 2022-10-11:
Guys, something like this:

https://youtu.be/jD05KqG-c6w?t=400

You can select a lot of options for every timer, this is really cool.

Could we get this into 2.9?