EdgeTX/edgetx #1893: The time on channel monitor display does not advance

rotorman, 2022-04-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When opening the channel monitor, the time is displayed in upper right corner, but it remains displaying the time, when the channel monitor was opened and does not advance.

### Expected Behavior

Current time should be shown

### Steps To Reproduce

Open Channel Monitor, note the time, have a couple of minute break and note the time again.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

Tested with 2.8.0-dev.

pfeerick, 2022-04-25:
Welp... this even even specific to the channel monitor... it looks like outside of the main (home?) screen, the time only updates when there is some other change to the screen... it doesn't refresh/update on it's own.

philmoz, 2023-03-02:
This was fixed in commit dca52c4ae