EdgeTX/edgetx #1894: Special function "Play Sound" does not play

kristjanbjarni, 2022-04-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Any special function using "Play Sound" does not play any sound.

### Expected Behavior

The specified sound should play.

### Steps To Reproduce

Go to special functions for any model and add special function:
Switch: ON
Func: Play Sound
Value: Beep1
Repeat: 1s

### Version

2.7.0

### Transmitter

Radiomaster TX16S

MervDale, 2022-04-25:
The problem may lie with your setup, SD card ect  I would suggest as the play sound works as it always has since the beggining.

kristjanbjarni, 2022-04-25:
I investigated this further and I found out that the special function "Play Sound" is confusingly directly linked to Sound Mode in radio setup. If you set this to "Quiet" or "Alarm" then play sound doesn't work at all. So is this really how it should work? I thought this setting was only related to sounds made by the radio itself related to telemetry and other system functions and not special functions set by the user.

pfeerick, 2022-04-25:
That's weird... "Play Sound" ... it's beeping for me. (also TX16S, v2.7.0) And you've checked volume settings (main AND beep volume) are not turned down? There is no need to check the sound card - Play Sound uses generated sounds, not wave files. 

[image]
[image]

pfeerick, 2022-04-25:
Ah, ok, well that's different... IIRC... Quiet should mean no sounds at all, Alarms is no sounds but alarms, NoKey is basically All without the keypress beeps, and then there is All.

kristjanbjarni, 2022-04-25:
Well I guess this is then behaving as designed, just a bit confusing that "Alarm" which was the setting I had is only radio alarms and not user alarms.

closed 2022-04-25 by kristjanbjarni

pfeerick, 2022-04-25:
Yes, for that to happen, there would need to be an option for user set alarms, rather than just to play a sound... maybe that could be another option for a special function play sound - i.e. the sound type. "Normal / Alarm"... which would also change the priority of it being played.