EdgeTX/edgetx #1896: Group and/or name logical switches and special functions

Daannoa, 2022-04-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I use logical switches to perform different tasks.
Each task may require several logic switches.
It then becomes difficult to know which logical switches are used for which task.
The situation is the same for the special functions

### Describe the solution you'd like

Would it be possible to add the ability to group logical switches ? ( same question for the special fonctions .)

this gives us the ability to group logical switches that are participating in the same task.

### Describe alternatives you've considered

An alternative would be to allow naming of logic switches and special functions.
(like INPUTS with their "Input name")

NemesisXB, 2022-04-25:
I like this. I would take it one step further in saying that you should be able to group logical switches, special funtions and global variables, as I have functions that require all three of these to accomplish the required task.

pfeerick, 2022-04-26:
I think a combination of both are needed... SF/LS need to be named and also comments/notes for each, and also grouping.

elecpower, 2022-04-28:
Whilst I am in support remember our B&W targets memory limits

pfeerick, 2022-04-28:
Yeah, B&W may be unfortunate one and not get this... due to memory AND firmware size constraints...

Neutral67fr, 2023-05-24:
There are many lines available for LS and SF... So I use to group the lines by functions, starting each line multiple of 10.
 e.g. LS01 to LS05 are for FM1 to FM5, LS10 to LS19 (or more if needed) are for SF triggering, etc...
  It's maybe not so convenient than names, but offer a standard organization (that allows copy/paste between models). But anyway naming limited to 3 char. is already a hassle for everything else (inputs, mixes, GV, curves...) so its doesn't help much anyway ;-)