EdgeTX/edgetx #1897: Emergency Mode when clearing model category list entry

mha1, 2022-04-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radio enters Emergency Mode when clearing model category list entry. Note: Firmware simulator crashes also.

### Expected Behavior

cleared model category list entry, no EM

### Steps To Reproduce

Hit Select Model from main menu
Hit the Model Categories list entry 
Hit Create category (new category named New shows)
Touch the new list entry and add one character a (now entry is Newa)
Press and hold scroll wheel (Edit, Clear popup menu shows)
Hit Clear --> Radio goes into EM, firmware simulator crashes

[image]

### Version

2.7.0

### Transmitter

Radiomaster TX16S

### Anything else?

This might be related to https://github.com/EdgeTX/edgetx/issues/1666  and  https://discord.com/channels/839849772864503828/839856795781431317/968216420716519494

pfeerick, 2022-04-26:
Perfect! That failed for me... so hopefully can finally put this to rest. I was sure there was an open issue for this still ... but maybe it was #1666 :confused:

eshifri, 2022-04-28:
Does not happen to me in the current 2.8 neither on TX nor in simulator.

mha1, 2022-04-28:
have you tried 2.7.0?

pfeerick, 2022-04-28:
Following those steps precisely I can reproduce this on TX16S with a recent dev... instacrash on clear. I'll have my normal debug TX16S on the debugger in the next day or so will be able to see what naughy stuff it does. 

Whatever it is... it is specific to the category just having been created, and being cleared... if you do other things like clear an already created category it is fine. I have this suspicion without looking at the code that since clear seems to be not just clearing the name, but changed it to a single blank space it is probably trying to clear the category name before it has been initalised type affair...

mha1, 2022-04-28:
there might be ways to crash it without clearing: https://discord.com/channels/839849772864503828/839856795781431317/968216420716519494

eshifri, 2022-04-28:
Now I can reproduce it. The thing I was missing is that "clear" should be done while editing the name.

eshifri, 2022-04-29:
The problem is caused by update():
the following fixies it, although I do not really understand why :-)  
          if(category->name[0] != ' ')
            update();

eshifri, 2022-04-29:
I think I understand what is happening there.
Long press on Enter sends two signals: KEY_BREAK(KEY_ENTER) and KEY_LONG_PRESS(KEY_ENTER).
First event causes the call to the changeHandler() which, in turn, calls update().
The second event calls the same changeHandler() of the same TextEdit.
But this TextEdit does not exist any more - update() caused complete re-build of the screen.
Hence the crash.
If we remove update() inside the changeHandler() everything works.

BTW, there is an unrelated bug in the line 752:
new TextButton(window, grid.getFieldSlot(3,2),TR_DELETE, [=]() -> uint8_t {
TR_DELETE instead of STR_DELETE.

mha1, 2022-04-30:
Awesome, thanks for digging deep.

Do you think the line 752 bug could be the cause for https://discord.com/channels/839849772864503828/839856795781431317/968216420716519494? As far as I understand there was no long press on enter involved.

eshifri, 2022-04-30:
This accident looks strange, to be honest. I doubt my explanation works for it if the description is accurate. The only thing I can imagine is something was pressed inadvertently. In this code any two events sent close to each other may cause a crash.
Of course if my understanding is correct.

mha1, 2022-04-30:
Strange indeed and unfortunately without a precise description to reproduce. Same goes for https://github.com/EdgeTX/edgetx/issues/1666 which was closed but didn't have a specific fix against it. The video would have helped. Both reports leave the feeling there might be lurking some other bug.

mha1, 2022-05-01:
Hi eshifri, thanks for looking into this and the PR to solve it. 

Please let me ask a question about key events. Why would a long press send two events at all? Isn't that in itself a bug with the way key presses are processed? I mean either it is a short press (key released before long press timer expired) or it is a long press (key kept pressed and long press timer expired).

eshifri, 2022-05-01:
Hi @mha1, 
I think the problem is in our attempt to utilize very small number of available controls as fully as possible.
When the key is pressed, several events are generated: the key was pressed; the key was released, the press may be long.
All events are processed independently in the onEvent() function.
Now imagine, that we edit the line and make some changes. We want to preserve the changes and the event, corresponding to key release must be processed. But it turns out that the press was long and we need to open the menu Edit (Clear, Copy, Paste).
We cannot ignore neither of these events, since user can close the menu without doing anything and in this case we want to keep the changes, performed by releasing the key. So this is not a bug. At least not an obvious bug.
It limits the possible content of the changeHandler() unfortunately. This is ugly, but this screen is the only place in the whole system where the update() is called from within this function. So...
BTW, if your cursor is not at the end of the string everything works fine, since changeHandler() does not need to be called.

The biggest problem here is the TextEdit class is used in many places and changing it does not look like a valid option: something may be broken in another screen.

PR #1925 (2022-05-04): Fix crash in category Clear (#1897)
Fixes #1897 

Summary of changes: Removed a call to update() in the changeHandler() of TextEdit.
One cannot call update()/rebuild() from TextEdit.changeHandler().
The reason is long press on ENTER sends two signals: KEY_ENTER and LONG_PRESS.
Both call changeHandler(). The first call causes update() and the screen is build().
So when te second call to changeHandler() comes the corresponding TextEdit is already deleted.
closed 2022-05-04 by pfeerick