EdgeTX/edgetx #1898: Cannot change to a different model 2.7.0 T-Pro ELRS

sk8board, 2022-04-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating the T-Pro ELRS to EdgeTX 2.7.0, I cannot change to a different model. When I select a different model, there is a pop-up screen that states, "Loading model...", then the screen returns to the original model. 

When creating a new model, I get the same result.

If I down-grade the firmware to the firmware that came with the T-Pro, the problem is not present.

I also have a T-Pro 4in1, which does not have this problem when upgrading to EdgeTX 2.7.0.

The problem is only present on the ELRS version of the T-Pro.

### Expected Behavior

When selecting a new or different model, I expect to change to the model I selected.

### Steps To Reproduce

1) Upgrade a T-Pro ELRS version to EdgeTX 2.7.0
2) Select a new or different model

### Version

2.7.0

### Transmitter

Jumper T-Pro

sk8board, 2022-09-05:
@pfeerick this bug does not seem to have any attention. Does the team need anything more?

pfeerick, 2022-09-05:
Have you tried updating to 2.7.1, or a nightly build to see if it has been resolved?

raphaelcoeffic, 2022-09-06:
> I tried the latest nightly build today and the problem is still present.

Thx for trying. We will need your settings/models to be able to reproduce.

pfeerick, 2022-09-06:
In other words, can you zip up a copy of the `MODELS` and `RADIO` folders from the radio that's playing up. I can't reproduce this on my ELRS T-Pro... it correctly flashes up the Loading screen, and the asterisk moves from the last model to the one selected or loading. And when I back out of the select menu, the correct model is loaded. So perhaps a model file is corrupt or something specific in the config is making it fail.

sk8board, 2022-09-06:
[Archive.zip](https://github.com/EdgeTX/edgetx/files/9500060/Archive.zip)

I have attached the `Models` and `Radio` folders as a zip file as you requested.

sk8board, 2022-09-25:
[Archive.zip](https://github.com/EdgeTX/edgetx/files/9500060/Archive.zip)

@raphaelcoeffic  @pfeerick  I have attached the `Models` and `Radio` folders as a zip file as you requested.

pfeerick, 2022-09-26:
Thanks for the reminder 😮 🙏 

Right, so I'm juggling a few things atm, and am still puzzled as to why the semver version field is zero instead of 2.7.0, 2.8.0 etc, but that should be a cosmetic issue. What is a problem is not that it won't change models... but after doing the first model switch (as I was able to switch models exactly one time with that file set on the radio) ... it would show the loading screen as you mention, freeze, and it would trigger an emergency mode reboot... as signified by a ! shown at the top right of the screen. 

What seems to be the cuplrit is that the trainer mode for all of the models is set to `Master/SBUS` ... when I turned that to OFF, on a model by model basis, I was able to switch models. I.e. turn it of, switch model. Turn it off for that model, switch model. And it seems to be fine after that. If a create a *new* model from scratch, i can switch too and from it without issue. But if I turn Master/SBUS on for that model, same deal. So for now if you at least make that change, you'll be right to go again. We probably need to check the hardware definitions as something is probably misconfigured re: SBUS on that radio, or it shouldn't be an option but isn't being sanity checked.

sk8board, 2022-09-26:
@pfeerick Thank you for checking into this.

I use Master/SBUS trainer mode all the time because I only fly with my son who is still training. So, I will have to stick with the original firmware that came with my T-Pro ELRS.

Please let me know when a fix is available so I can update to newer firmware.

pfeerick, 2022-09-27:
Hm... so how do you use SBUS with the T-Pro?

sk8board, 2022-09-27:
@pfeerick  @raphaelcoeffic  I was able to change trainer mode to `Master/CPPM` as a workaround for this problem. I have verified this workaround works very well.

I know the ELRS team will not support SBUS or PPM for receivers, which is why I created this enhancement request for trainer mode.  #https://github.com/EdgeTX/edgetx/issues/1607  

**_If enhancement request #https://github.com/EdgeTX/edgetx/issues/1607   cannot be supported_**, then it is important for the T-Pro and other radios to support SBUS trainer mode. This is because no one makes an ELRS to PPM converter. Matek makes an ELRS to SBUS converter which can be used with an ELRS receiver and SBUS trainer mode for wireless trainer function. 

**_If enhancement request #https://github.com/EdgeTX/edgetx/issues/1607   can be supported_** and the EDGE team does not want to support SBUS on the T-Pro, then please remove `Master/SBUS` from the menu so people know to use CPPM rather than chase a problem related to SBUS that will not get resolved.

I appreciate your help!

sk8board, 2022-09-27:
@pfeerick @raphaelcoeffic 

I made a video to better explain my writing above.

[Here is the link to the video](https://youtu.be/lMYfBj8jtrM)

pfeerick, 2022-09-27:
Yes, that video clearly explains what you are after, thanks! :) I was more interested in the *how* it was connected, as I wasn't aware of any documentation on SBUS being a supported function on that transmitter. I noticed it doesn't crash if you re-enable Master/SBUS afterwards, so does it actually work then? i.e. It only crashes when changing models, but otherwise functions? Just means we are getting closer to identifying exactly where it is going wrong.

re: CRSF trainer input, did you give the PR that was linked to that issue a try to see if the works? https://github.com/EdgeTX/edgetx/pull/1462 It will need some re-work as it needs to fit in with the changes to the serial model, but I did have CRSF student input working on a TX16S via the module bay.

sk8board, 2022-10-01:
@pfeerick 

I tried to use trainer mode `Master/SBUS` with 2.8.0-rc1. The result was erratic channel operation when trainer mode was enabled. 

When using the original T-Pro ELRS firmware, `Master/SBUS` trainer mode worked as expected. Maybe the SBUS code from the original T-Pro ELRS firmware can be used?

> re: CRSF trainer input, did you give the PR that was linked to that issue a try to see if the works? https://github.com/EdgeTX/edgetx/pull/1462 It will need some re-work as it needs to fit in with [...]

With regard to testing a PR, I assume you mean to compile the PR for use? If so, I am not savvy enough to compile.

pfeerick, 2022-10-01:
AFAIK, nothing was changed in the SBUS code. Note likely it is some clash/incompatability with the lower level serial driver rewrite and the tpro configuration - which wouldn't have been picked as SBUS isn't exactly plug and play on that radio ;) I'm on phone atm, but will link the guide for self-compile... It's pretty easy now :)

pfeerick, 2022-10-12:
Would you mind checking everything is working ok now? You can either use the next nightly, or 2.8.0-RC3 which is likely to come out later today.

closed 2022-10-12 by pfeerick

sk8board, 2022-10-13:
> @sk8board Would you mind checking everything is working ok now? You can either use the next nightly, or 2.8.0-RC3 which is likely to come out later today.

@pfeerick  Yes, I will test 2.8.0-rc3.

sk8board, 2022-10-13:
@pfeerick I have tested 2.8.0-rc3 and verified this problem is fixed.

Thank you to all who solved this problem.

pfeerick, 2022-10-14:
Fantastic! :)

PR #2515 (2022-10-12): fix: prevent endless loop with SBUS trainer on model change
## Description

De-initialising the external module while SBUS trainer is used on external module RX leads to the following sequence:
- `extmoduleStop()` is called, thus shutting down the timer DMA
- on certain targets, the timer DMA stream is the same as the external module RX DMA stream
- when this DMA stream is used by the SBUS trainer input, this leads to DMAFifo to return random bytes continuously
- thus driving `processSbusInput()` into an endless loop, which is run in the mixer task
- as the mixer task has the highest priority, [... cut]
commit 655965d7 (2022-10-12): fix: prevent endless loop with SBUS trainer on model change (#2515)

* fix: prevent generic module de-init if protocol is NONE

Fixes #1898

De-initialising the external module while SBUS trainer is used on external module RX leads to the following sequence:
- `extmoduleStop()` is called, thus shutting down the timer DMA
- on certain targets, the timer DMA stream is the same as the external module RX DMA stream
- when this DMA stream is used by the SBUS trainer input, this leads to DMAFifo to return random bytes continuously
- thus driving `processSbusInput()` into an endless loop, which is ru [... cut]
commit 0371814f (2022-10-12): fix: prevent endless loop with SBUS trainer on model change (#2515)

* fix: prevent generic module de-init if protocol is NONE

Fixes #1898

De-initialising the external module while SBUS trainer is used on external module RX leads to the following sequence:
- `extmoduleStop()` is called, thus shutting down the timer DMA
- on certain targets, the timer DMA stream is the same as the external module RX DMA stream
- when this DMA stream is used by the SBUS trainer input, this leads to DMAFifo to return random bytes continuously
- thus driving `processSbusInput()` into an endless loop, which is ru [... cut]