EdgeTX/edgetx #1899: Zorro with XJT Lite won’t enter bind mode

rdockstader, 2022-04-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a Radiomaster Zorro ELRS with an FrSky XJT lite module. I’ve set up a new model, disabled internal, and set External to XJT lite, D8. However when I press the bind button, the external module fails to enter bind mode. The indicator light doesn’t show it’s entered bind mode, and no receivers will bind to it. 

### Expected Behavior

External module should enter bind mode and bind to the quad. 

### Steps To Reproduce

1. Create a new model
2. Disable internal module
3. Enable external module, set to XJT Lite, D8
4. But the radio in bind mode. 

### Version

2.7.0

### Transmitter

Radiomaster Zorro

pfeerick, 2022-04-26:
A workaround seems to be to set the module to protocol to D8, then highlight XJTLite, and then select it again, since as soon as you select the module, it gets power-cycled, and will be in the right mode ;)

I can reproduce this with Zorro and XJT Lite. Doing module functionality tests only (i.e. not link test) when set to D16 module lights indicate bind, range check and normal modes correctly. However, setting to D8 or LR12 and it does not do anything, and when setting it back to D16 the module does not bind or range check any more. 

It would appear that when changing modes the module is not power cycling? I noticed this when doing comparison test on OpenTX (which does appear to be working properly when changing modes). OTX appears to be power cycling the module, whereas ETX does not. 

A secondary bug seems to be that sometimes the transmitter will revert to XJT (`TYPE_XJT_PXX1`) rather than XJT Lite (`TYPE_XJT_LITE_PXX2`) - haven't been able to determine exact sequence to reproduce yet - seemed to somewhat be related to going from D8 back to D16.

pfeerick, 2022-04-26:
It looks like following the rabbit hole of why was XJT Lite being changed to XJT lead to the culprit :)

rdockstader, 2022-04-26:
Sounds like you're already well down the path on this one, but I did want to confirm that your workaround works for me as well :)

PR #1904 (2022-04-27): fix: XJT Lite module type being replaced with XJT
Fixes #1899

#1702 fixed an issue where external module setting was overwriting internal modules. I didn't take into account that `isModuleXJT()` doesn't descriminate between XJT and XJT Lite module, so it obliterates the XJT Lite type on the first change made. In doing so, the XJT Lite was not power cycled/mode changed.
commit 59577455 (2022-04-27): fix: XJT Lite module type being replaced with XJT (#1904)

Fixes #1899
commit 20030f74 (2022-04-27): fix: XJT Lite module type being replaced with XJT (#1904)

Fixes #1899
closed 2022-04-27 by pfeerick