EdgeTX/edgetx #1905: Bluetooth Discover dialog needs vertical scroll bar addition

rotorman, 2022-04-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With Bluetooth built into firmware, hitting the discover and finding in vicinity more Bluetooth devices than fits the screen, it is luck if the device one wishes to connect to is in the list or if it is outside the window scope and not selectable.

@dlktdr made a video showing this: https://photos.app.goo.gl/WGGY4mQop9BLdfVV9

### Expected Behavior

The Bluetooth discover dialog should show a vertical scroll bar to allow displaying/selecting the Bluetooth partner to connect to, also if more Bluetooth devices will be discovered than fits the screen.

### Steps To Reproduce

Build firmware with -DBLUETOOTH=YES, use either FrSky PARA module or ESP32 flashed with https://github.com/dlktdr/BTWifiModule enable Master/Bluetooth trainer. But more than 6 discoverable Bluetooth devices in the vicinity, press Init and then press Discover - only up to 6 will be selectable.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

2.8.0-dev

gagarinlg, 2023-03-18:
@rotorman is this still a problem in main?

rotorman, 2023-03-19:
The situation seems to be even worse with -DBLUETOOTH=YES, as just tested with main branch and TX16S with FrSky BT CC2540 module - there is no successful communication with the module :(