EdgeTX/edgetx #1907: TBS Agent Lite

hickscorp, 2022-04-26:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am fairly new to the whole RC hoby, however I think I found a bug with EdgeTX.

I use the Tracer module, and it works great using OpenTX, including the TBS Agent Lite scripts.

But when I install it on EdgeTX:
- The script is not visible in the `System > Tools` section.
- Using the SD file Explorer and long-pressing the scroll wheel and choosing `Execute`, it says that there's an error line 8, that it cannot call a ` nil` value.

### Describe the solution you'd like

Ideally, TBS Agent Lite shall run without monkey patching it, so that the LUA API is future proofed on EdgeTX.
But if someone proposes a monkey patch I'm happy to run it and report, if it helps.

### Additional context

I am using a RadioMaster TX-12, along with the latest EdgeTX build.

It is difficult for me to look at the call stack, because it seems that TBS Agent Lite comes precompiled (`luac` files instead of `lua`) so I have very limited visibility on the root cause of the problem.

pfeerick, 2022-04-27:
If it's not in the tools section, that suggests it has been mis-installed - as it should come up in SYS -> Tools when installed in the correct location. 

i.e Is this what you did (adapted from [TBS install page](https://www.team-blacksheep.com/products/prod:agentx))
>    Download newest version of Agent (0.96 at time of writing)
>    Unpack downloaded ZIP file
>    Move included files and folders to the /SCRIPTS/TOOLS/

And yes, you will have trouble making any sense of it as it comes pre-compiled rather than as source code. 

@JimB40 Your thoughts? :)

hickscorp, 2022-04-27:
It is possible that I put all the scripts directly at the `/SCRIPTS/` folder... I will confirm today when I have access to the transmitter.
If that's the case, I guess we will close this issue and flag it as "user error" ;)

closed 2022-04-27 by hickscorp

pfeerick, 2022-04-27:
No problem, glad it was that simple! Happy flying! :)

hickscorp, 2022-04-27:
Thanks a lot Peter! Happy flying to you too.