EdgeTX/edgetx #1909: Jumper T-Pro S1/S2 pots resolve to wrong values on 2.6/2.7

gamlebardun, 2022-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

a debug readout of S1/S2 gives 0-2048 (well, 0-2046), but the value returned for use in input/mix is only the part from 0-100, so the action of the rollers/pots is only the end 5% or so. A value > 100 just returns 100.
I'm not sure when this was introduced, my radio is quite new, but I updated to 2.6 when I received it, and when I saw this issue I updated to 2.7 - same behaviour.

### Expected Behavior

as the pot jogs between 0-2048 ADC-value, the returned value is scaled  -100 - 100 

### Steps To Reproduce

1. Add an input or mix from S1 (or S2)
2. Observe output as you jog the wheel from one endpoint to the other

### Version

2.7.0

### Transmitter

Jumper T-Pro

### Anything else?

[image]

Look at the values of the A5 ADC-channel for different input values

pfeerick, 2022-04-28:
Have you tried doing calibration again (SYS -> Hardware -> calibration? As you need to center and move all analog inputs during calibration (so sticks + S1/S2 on TPro). If you don't, weird things will happen in the raw sampling -> 100 - -100 conversion. If you are running the full range of 2048 (actually 4096 - check the second page of the analogs screen for the raw values) then it should just be a calibration error. 

Slider movement
https://youtu.be/mLlO8gdlYUE

Diagnostics screen
https://youtu.be/oHXZT_LfpRY

gamlebardun, 2022-04-28:
Oh, this is embarrasing .. I did not see the calibration included the pots as well. Thank you, in high embarrasment mode!

closed 2022-04-28 by gamlebardun

pfeerick, 2022-04-28:
No problem... this is something that needs to be corrected... for both colorlcd and B&W, as it isn't clearly pointed out ;) The bars *are* there for pots and sliders, but naturally your focus is on the sticks, not any other analog inputs.

gamlebardun, 2022-04-28:
Yes, I expect you are right. Well, thanks again, @pfeerick