EdgeTX/edgetx #1911: Zorro simulator window is missing a screenshot icon

rotorman, 2022-04-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

No simulator icon shown for Zorro (tested with Windows 64-bit simulator):

[image]

### Expected Behavior

Screenshot icon, such as here for TX16S should be there also for Zorro:

[image]

### Steps To Reproduce

Open RadioMaster Zorro firmware simulator in Windows.

### Version

Other (Please specify below)

### Transmitter

Radiomaster Zorro

### Anything else?

2.8.0-dev

pfeerick, 2022-04-28:
Depending on how far you want to go with that list, also applies to Jumper T-Lite and T-Pro, RM TX12, Flysky NV14 (which I will give a pass anyway as we also haven't figured out the trim hats yet either 😁 )

elecpower, 2022-04-28:
If toolbar visible can use screenshot icon even if no icon on radio image

pfeerick, 2022-04-28:
Shortcut/hotkey also... I wonder if it would be better to remove the screenshot icon, since the menu bar / hotkey are there (for consistency). Or just leave it as wont-fix as it really isn't necessary.

elecpower, 2022-04-28:
For consistency I favour removing radio icon and saves having to code the functionality into every radio sim.

gagarinlg, 2025-01-24:
@elecpower can this be closed?

elecpower, 2025-01-24:
Fixed by #5692

closed 2025-01-24 by elecpower