EdgeTX/edgetx #1916: HM ELRS 2.4 fail on EdgeTx 2.7 TX16s

domhan, 2022-04-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Update from 2.6 to 2.7. The HM ELRS did not work. The LED on the module lit. I have tried different Baud-rates - no success. No connection/telemetry from the model, no chance to use the ELRS Lua.Script.

I have done a downgrade to 2.6 and all worked fine. Tried the update to 2.7 again, same problem.

### Expected Behavior

I have not changed the configuration and Version of the HM ELRS-Modul. So I had expected, that it should connect without any problem.

### Steps To Reproduce

See description.

### Version

2.7.0

### Transmitter

Radiomaster TX16S

### Anything else?

If I can help with debugging, I need some description.

mituritsyn, 2022-06-02:
seems something is broken in config. I've fixed by next steps:
- downgrade to 2.6 
- configure elst to workink state - some settings need to be switched. i've switched onebit+bod rate + crsf in model to reinit module
- update to 2.7 and have 1 starting error with 400k 250Hz

domhan, 2022-06-02:
I have made a "Simulator-Model" without any internal or external Module. After switchen to this Model and switching back to my ELRS-Model - all works fine. It seems, there is an "autonegotiation" or "initialisation", which fixes the error.
I'm on EgeTX 2.71 an ELRS 2.50 on the TX16s.

mituritsyn, 2022-06-02:
> seems something is broken in config. I've fixed by next steps:
> 
> * downgrade to 2.6
> * configure elst to workink state - some settings need to be switched. i've switched onebit+bod rate + crsf in model to reinit module
> * update to 2.7 and have 1 starting error with 400k 250Hz
Sorry, rushed it, after reboot the errors again.
Switching to sim model doesn't work too

gagarinlg, 2023-02-25:
Does this also happen with 2.8?

mituritsyn, 2023-03-02:
> Does this also happen with 2.8?

I've added tbs inverter so I can't provide info about initial case, sorry

closed 2023-09-05 by raphaelcoeffic