EdgeTX/edgetx #1917: Opening TBS Agent Lite Wifi setting hangs the radio

flyhigh-hifi, 2022-04-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have working(slowly but working) wifi system to connect phone with MicroTX V2 module on OpenTX, now if I try to go to Agent Lite lua script(v0.96) and open XF Wifi settings - whole radio hangs(sticks are working but no way to change the screen - holding exist doesn't help). I'm not sure whether issue is in tbs lua script or edgeTX, but it shouldn't hang at least.

### Expected Behavior

Expecting to have working tbs agent lite.

### Steps To Reproduce

1. Install tbs agent lite 0.96
2. bind radio
3. open lua script(XF Wifi setting)
4. ...

### Version

2.7.0

### Transmitter

FrSky X9D+2019

JimB40, 2022-04-30:
Is your Crossfire module WiFi firmware version above 2.0?

flyhigh-hifi, 2022-04-30:
> Is your Crossfire module WiFi firmware version above 2.0?

Forgot to mention, yes, latest one v6.17 firmware

JimB40, 2022-05-08:
Crossfire firmware and WiFi module inside transmitter are two different firmwares.
So you can have 6.17 firmware for Crossfire but still WiFi firmware is 1.XX which is incompatible with Agent Lite 0.96 and you need latest 2.XX
As now you can't chesk wifi firmware version via Agent Lite (it hangs) use prcedure described here:
https://team-blacksheep.freshdesk.com/support/solutions/articles/4000161703-tbs-crossfire-update-the-wifi-module-

athertop, 2022-05-14:
I am seeing similar issues with Agent Lite after choosing WiFi module and General menu option. It hangs after displaying the first line of the General screen when it opens up. There is a suggestion in RCG thread that this might be lua related (and specifically only started in recent release of ETx - bug not in current OTx), where background scripts are being allowed to run at same time as the Tools and telemetry scripts causing them to hand or re-initialise. This is the post from Alex/Yaapu: https://www.rcgroups.com/forums/showpost.php?p=49110363&postcount=2593

 I am seeing similar issues with the Yaapu scripts/widgets which continually try to re-initialise. I posted here: https://www.rcgroups.com/forums/showpost.php?p=49170383&postcount=840

athertop, 2022-05-16:
I should add, that it is reported that the lua issues seem to have emerged in only the latest releases of ETx. Earlier releases are reported as not having this issue in the Yaapu link above. Yaspu/Alex suggests that lua handling appears to have been changed in the newer releases of ETx, allowing background lua processes to run, which interrupts the working Lua script (yaapu telem screen or tbs agent lite etc in examples seen).

urlu75, 2022-05-16:
for ELRS it works with edge 2.6 but lua script is getting CSRF packet sooo slowly.
After official 2.6 it is not working any more. I'm wondering if there is a link with this [PR](https://github.com/EdgeTX/edgetx/pull/1498)

flyhigh-hifi, 2022-05-20:
> Crossfire firmware and WiFi module inside transmitter are two different firmwares. So you can have 6.17 firmware for Crossfire but still WiFi firmware is 1.XX which is incompatible with Agent Lite 0.96 and you need latest 2.XX As now you can't check wifi firmware version via Agent Lite (it hangs) use prcedure described here: https://team-blacksheep.freshdesk.com/support/solutions/articles/4000161703-tbs-crossfire-update-the-wifi-module-

Confirming that this fixed my TX being blocked by lua script and tbs script/wifi section works fine after update. Thanks.

athertop, 2022-05-20:
> Confirming that this fixed my TX being blocked by lua script and tbs script/wifi section works fine after update. Thanks.

But do you run the Yaapu telemetry script/widget? If so, display the yaapu widget screen then push sys and launch the tbs agent lite and see how it has issues because of conflicting use of telemetry.

JimB40, 2022-05-20:
Executing LUA script in stand-alone mode should stop all other scripts including widgets.
@jfrickmann have you been changing anything in this area?

jfrickmann, 2022-05-20:
No, not as far as I know!
————————
Jesper Frickmann

athertop, 2022-05-20:
It worked fine on 2.6 but seeing issues since 2.7 so something appears to have been introduced to cause this

jonas-koeritz, 2022-05-23:
Can confirm, running the Yaapu script will kill anything that uses crsf telemetry from the TOOLS. Either painfully slow or not working at all, as if they're competing for the crsf_pop_telemetry calls.

lai573, 2022-05-30:
I am running edgetx 2.6,but I still have the problem

athertop, 2022-05-30:
But then the problem reappeared in 2.7.1

Ltomi78, 2022-09-04:
I have exactly the same issue with TBS agent lite and yaapu script. With models running Yaapu widget at background TBS agent lite freezes during loading data for rx and does not find Micro TX as viewabl option. As i change to an other model without Yaapu widget setup...TBS agen lite works as it should.
Radiomaster tx16 edgetx 2.7.1
Yaapu 1.96 dev
TBS agent lite 0.96 
TBS FW 6.14 tx and rx
Any suggestion or new version for edgetx or yaapu which fixes this bug?

Ltomi78, 2022-09-04:
> For people with mk2 tx16s, it comes with edgetx 2.6,but it has this problem. I just reload the firmware 2.6 from this site. Turns out they are not the same because the problem is gone now.

I can confirm that my brand new tx16 mark2 comes with 2.6edgetx has this issue when yaapu selected for a certaim model tbs agent lite freeze or just not load the data

pfeerick, 2022-09-04:
Probably similar to two comments above, the version of edgetx shipped was probably **not** the release version of 2.6, but a nightly build. It's easy to check - what's the release date shown in Radio Settings => Version? Is it "2022-01-24" (2.6 was released on 24 Jan 2022). If not, you could try flashing the proper 2.6 release build. 

There are likely to be two different issues present here - as B&W radios (like the initial poster) are very different to colorlcd radios in how they and Lua work. For colorlcd radios, we should be stopping widgets when you're in the settings, but it appears possible we aren't atm, which is why the combination of yappu + TBS agent lite is problematic, as they are both trying to talk to the crossfire module at the same time.

Ltomi78, 2022-09-04:
Thanks for your feedback. I ve reflashed the stock supplied version of 2.6.0...its an early version from 2021 12.20..so not the one you mentioned. But know with this 2021 version of 2.6.0 the tbs agent lite started to working along with yaapu widget :) a little bit slow but finds micro TX also. I will check it with the version from 2022.01.24 how its behave.
Thx :)
But thats true with 2.7.1 the issue exist.

> Probably similar to two comments above, the version of edgetx shipped was probably **not** the release version of 2.6, but a nightly build. It's easy to check - what's the release date shown in [...]

Ltomi78, 2022-10-04:
Any news if this issue with tbs agent lite and other scripts using crsf telemetry at the same time will be fixed in 2.8.0?

athertop, 2022-10-04:
I would love to hear this answer as I'm finding it difficult using crossfire with EdgeTX. Could one of the devs please advise if this issue is now resolved? I do not see any recent commit for this

Ltomi78, 2022-10-04:
> I would love to hear this answer as I'm finding it difficult using crossfire with EdgeTX. Could one of the devs please advise if this issue is now resolved? I do not see any recent commit for this

Agree maybe it would be nice if @yaapu Alex can take part in the examination what causing this issue. Yaapu script works well but seems its overrides TBS Agent lite which freezing down.

athertop, 2022-10-04:
It's not the yaapu script specifically causing this. It would happen with any telemetry widget. It's that EdgeTX is not pausing the widget scripts when you open tools scripts, so both are fighting/conflicting over the telemetry feed. It works fine in Opentx and did in an earlier version of ETx also. So something has been introduced to cause this issue

Ltomi78, 2022-10-04:
> 

Agree thanks for clarifying...and i coud not force to stop that scrip without deleting it

athertop, 2022-10-05:
@flyhigh-hifi I wonder if it might be an idea to rename this issue to maybe get it better noticed? I was thinking something like "Tools scripts hang if telemetry widgets active". Just an idea as the issue does not only relate to TBS scripts

Ltomi78, 2022-10-09:
> @flyhigh-hifi I wonder if it might be an idea to rename this issue to maybe get it better noticed? I was thinking something like "Tools scripts hang if telemetry widgets active". Just an idea as the issue does not only relate to TBS scripts

Good idea!

raphaelcoeffic, 2022-10-10:
Gentlemen, has anyone tested this with the latest 2.8.0 release candidate? I believe the issue should be gone, as no method from the LUA widgets are called while in a standalone LUA script.

urlu75, 2022-10-10:
tested with 2.8RC2 rebuild with BT=ON and GPS=OFF. 
ELRSv2 script now perfectly run with yaapu set up.
Issue seems to be gone, thanks !

pfeerick, 2022-10-11:
I was going to say I'm not so sure... but I think there is something else afoot here - it could be that the yaapu telemetry widget is still being serviced when going into TBS AgentX lite, or there is some other timing weirdness - it seemed more apparent when I had just added the widget, than powering up with a configured model. i.e. I opened TBS Agent Lite, and it was blank, opened up ELRS and it was busy loading for about 15 seconds, and then all of a sudden started slowly pulling in the data and was fine after that. Went back to TBS Agent, and it was fine also. And upon restarting the radio, was generally immediately responsive. So I think we can safely call this basically fixed, but there could be some edge case weirdness to keep an eye out for (although I half expect another PR that is already waiting might have some impact on that).

closed 2022-10-11 by pfeerick

athertop, 2022-10-30:
I tested 2.8rc3 tonight with yaapu script and tbs agent lite. Working perfectly now. First time I have been able to use these properly since adopting ETx. Well done guys. Can't wait for the final release. Best regards

Ltomi78, 2022-10-30:
> 

Thanks for your feedback..cant wait to test it on my radio too