EdgeTX/edgetx #1918: model text info file

ror1948, 2022-04-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

it seems to work on some but not all
maybe in 2.5 it worked better since that was first move from opentx to edgetx

### Describe the solution you'd like

https://www.youtube.com/watch?v=HaQWqG8ehpA

eshifri, 2022-04-30:
OpenTX implementation is completely different.
You'll have to provide more details, since it works fine for me.
Please notice, that the name of the file must be one of the following:
1. ModelName.txt
2. modelXX.txt (XX is the number of the model file).
If ModelName.txt in ETX may contain spaces. OTX version, where space has to be substituted by "_" is also supported, but on TXs with color LCD only.

ror1948, 2023-01-12:
Now on EdgeTx2.8 and it does not seem to be working as described.
Manual says if "display checklist" in model setup is on then the model info file should be displayed after you select the model.
It does not display the file it just sits there in the select model screen. txt file is displayed if I select "model notes"

eshifri, 2023-01-12:
This is fixed by #2922 and it is not in 2.8.0
It will definitely be in 2.9.0; not sure about 2.8.1

pfeerick, 2023-01-12:
Yup, it's (#2922) in the 2.8 branch (was milestoned for 2.8.1) and it shouldn't be too long before that is released now.

closed 2023-01-12 by pfeerick