EdgeTX/edgetx #1921: Discrepancy between simulator and Radiomaster zorro

SadE54, 2022-04-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Hi , 

I set up a throttle arming system using a gloval var , logical switches , and specials functions
[image]
[image]
[image]
s

It's working well on the simulator , and behaves accordingly logical switch states.  But on the zorro if it's arming , the 'armed' track is played, then just after the 'disarmed' is played too. And in the oyer way it's the same ('armed' is played' just after 'disarmed'

### Expected Behavior

The output behavior is correct , the only problem is a second second file is played only on the radio , not the simulator.

### Steps To Reproduce

1. Thr is set to low
2. Press for 2 seconds on SA and release
3. 

### Version

2.7.0

### Transmitter

Radiomaster Zorro

SadE54, 2022-05-01:
Someone told me to put a 0.1s delay on L06 and L07 and it's working well now.
I guess it could be related to switch bounces on the Zorro ? There's no input filtering at lower level ?

raphaelcoeffic, 2022-05-01:
> There's no input filtering at lower level ?

At hardware level: simple RC filter, as usual.

SadE54, 2022-05-02:
Maybe it could be a good idea to software debounce to avoid false triggering? I don't know the polling time, but just validate level when 2 following read values are the same should be enough

pfeerick, 2026-09-30:
> Someone told me to put a 0.1s delay on L06 and L07 and it's working well now.

This is the correct solution. Switch debouncing is not the issue, but rather the fact that there is a feedback loop due to how logical switches, special functions and edge detection works... 

When you release SA after holding it for 2 seconds, the Edge switch (L02) stays on for about 0.1 seconds, not just an instant. During that time the radio checks the logic about 10 times.

Your arm state (GV1) is both the result of the toggle and part of what triggers it:

1. GV1 is 0, so L03 turns on. The radio sets GV1 to 1 and plays "armed".
2. A moment later L07 sees GV1 = 1, and L02 is still on, so L04 turns on. The radio sets GV1 back to 0 and queues "disarm".
3. This keeps flipping back and forth until L02 turns off.

With the Zorro's timing it happens to finish on the right value, which is why the arm state looks correct. You still hear both sounds, because "disarm" was queued behind "armed" during the flipping.

The simulator runs its logic much less often (about every 50ms), so only one flip fits in that 0.1 second window and you hear just one sound. This is just the nature of how things run differently on a dedicated microcontroller vs a general purpose PC that's running hundreds of other instructions.

Your 0.1s delay on L06 and L07 is the right fix. After GV1 changes, the other side can't turn on until L02 has already switched off, so the toggle only happens once. Keep the delay at 0.1s or more.

closed 2026-09-30 by pfeerick