EdgeTX/edgetx #1926: Timers count up option?

Daannoa, 2022-05-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently if we do not define a "start" time in the timers they count upward.
If we define a "start" time in the timers they count downward and prevent me when the time set in “start” is past.

Personally, when I fly, I don't find this count down intuitive (you have to do a subtraction to find out how much time has passed) 
But I like to be notified when the time is up.

### Describe the solution you'd like

Would it be possible to have an option that would allow us to define in which direction the timer counts? (Count up / count down)

This would allow us to have a more intuitive timer AND be notified when the time is up.

### Describe alternatives you've considered

Currently I'm using logical switches to get the job done, but it's also not intuitive to set the "timer time" in the logical switch.

eshifri, 2022-05-01:
Just set the starting value for timer to zero.

pfeerick, 2022-05-01:
Title was a bit misleading @eshifri ... he is aware of setting it to zero, but wants to be able to count up to a time so don't have to do the math to figure out time elapsed.

Interesting idea... could be implemented with a choice (so you can can "Count direction: Up/Down", and then have the timer logic take that into consideration... zero for the time would then just mean no alarm if counting up.

eshifri, 2022-05-02:
Aha, now I see. Personally, I think  this is counterintuitive: time left is more interesting, than time elapsed. But why not? :-)

lshems, 2022-05-02:
Suggestion: leave all as it is, and just create a display option that shows timer setting minus remaining time.

Call the setting 'display' with options 'elapsed' and 'remaining'.

I do that in all my apps where I need an elapsed time, which is useful when you 'think' in target times that may vary.

lshems, 2022-05-02:
This would prevent creating two sets of logic which only differ by the reference frame.

Voice commands should follow displayed value, or do you want to be able to set that independent? Yet another variable. Lol.

pfeerick, 2022-05-03:
True, it is only the display logic that needs to change, not the actual timer logic - and there is no reason for the voice prompts to change - they only state the time :)

eshifri, 2022-05-07:
I have functionality implemented, but with UI and storage I will wait for lvgl coming in.

eshifri, 2022-05-10:
BTW, what do you think will be a better UI implementation: checkBox show elapsed or Choice: show Elapsed/show Left?

pfeerick, 2022-05-11:
IMO a choice of "Show Elapsed / Show Remaining" ... no guessing as to what it shows then ;)

eshifri, 2022-05-15:
Next question: color LCD only, or B/W radios as well?
I would think that for small screen radios this TAB is already overloaded with options...

elecpower, 2022-05-15:
IMO having functionality differences between radios should only be due to radio capability not because the ui would be cluttered. If we have differences in settings then we have to add conversion code too and deal with the ongoing why doesn't this work or setting 'x' is missing on radio 'y'

MervDale, 2022-06-13:
What is the difference with having the timer count up as it will announce the minutes, if you want to fly for six minutes then start your landing procedure at the five minute call.  I use the countdown timer every flight and just reset timer before every flight or it can be reset at engine off switch. Total flight time can be kept on another timer and manually rest if wanted

eshifri, 2022-06-13:
The difference is the timer will expire, change color and announce expiration.
Personally, I doubt I would ever use it, but somebody finds it convenient - why not?

lshems, 2022-06-13:
In competition flying you sometimes need to fly a certain time several times within a working window. Counting down to what you still need to fly is then for some much more intuitive to focus on then how much you have flown already. Also, it means that the countdown to do a quick turnaround for the next flight will always be the same countdown to zero. This will lessen the thinking, certainly on targets that are not whole minutes.

PR #2038 (2022-08-02): Select timer count direction (#1926)
This PR allows user to select timer direction: display/announce elapsed time or remaining time.

Fixes #1926

Summary of changes:  add UI for BW and color radio to select direction of the timer counter.
Support TX and Companion. 
Based on the main after lvgl merge.
closed 2022-08-02 by pfeerick