EdgeTX/edgetx #1927: Trainer port issue on Radiomaster TX16S

Avalonnw, 2022-05-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am having issues with trainer port on ETX2.7. Migrated from 2.5 on TX16S and head tracker stopped working. Switched back to 2.5 and all is working again (did not disconnect the cable even).
Tried 2.7 RC1 and RC2, as well as 2.8 nightly. Same result. No trainer port signal coming through.
I have bin files on flash card and reflash radio back and forth without touching the trainer cable.
I've noticed on Inputs page the inputs associated with TR5 and TR6 (the ones I use) are not highlighted in yellow and there is no curve shown on edit input page.
Also, in Mixers, if I set a switch to mix (replace) one channel on SA(UP) and set source to Input associated with trainer port, the switch works, HOWEVER if source is set to TR5 directly, the switch does nothing (not highlighting). Can make a video if my description does not make sense.

### Expected Behavior

Trainer port should work the same way as in v. 2.5

### Steps To Reproduce

As described above:
1. On Input page: one input set as TR5 as source. Call it HT1
2. On Mixers page: One channel source set to HT1. Switch set to SA(UP).
  Switching SA to UP makes mix highlighted but there is no valid output on the channel.
2a. On Mixers page: One channel source set to TR5. Switch set to SA(UP).
  Switching SA to UP does NOT make mix highlighted and there is no valid output on the channel.

Same steps in 2.5 work perfect.

### Version

2.7.0

### Transmitter

Radiomaster TX16S

Avalonnw, 2022-05-02:
My bad. Setting Trainer port to Master helped. Not obvious considering 2.5 worked without it.

closed 2022-05-02 by Avalonnw

elecpower, 2022-05-02:
Defaulted to Master at 2.5 however OFF was added for consistency across radios