EdgeTX/edgetx #1928: Sequence of Trims in flight modes menu not obvious

mha1, 2022-05-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

[image]

The order of Trims to be manipulated can not be identified by looking at the Trims entry. This is frustrating and forces the use companion to adjust the trims behavior.

According to https://doc.open-tx.org/opentx-taranis-manual/software-overview/model_menus/flight_modes the order is RETA and most likely T5 and T6. This is an arbitrary order and only somewhat intuitive for a single control mode.  And who reads manuals, right?

### Describe the solution you'd like

If there is space enough the simplest solution might be to add the character representing the control that's manipulated with each entry, eg:
R=0, E+2, T=0, A=0, T5=0, T6=0 instead of =0, +2, =0, =0, =0, =0 

### Describe alternatives you've considered

Alternative might be a headline but this would limit the number of flight modes on screen to two. I could live with that in favor of clarity.

pfeerick, 2022-05-03:
Duplicate of https://github.com/EdgeTX/edgetx/issues/1788

mha1, 2022-05-03:
You want me to close it in favor of https://github.com/EdgeTX/edgetx/issues/1788?

pfeerick, 2022-05-03:
Yeah, I think it is safe to do so... as mentioned in that one this is something likely to be fixed as part of the LVGL UI re-write. and there was some mention of exactly this the other night (but for performance reasons), so one way or another this will happen. Regardless, this adds to that issue since it's cross-referenced. 😉

mha1, 2022-05-03:
Closed in favor of https://github.com/EdgeTX/edgetx/issues/1788

closed 2022-05-03 by mha1