EdgeTX/edgetx #1929: Copy/Paste shouldn't try to overwrite existing model

RJMcInty, 2022-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When pasting a model into a model and settings file, it will always attempt to overwrite the selected model, rather than pasting it as a new model, or prompting for overwrite if the model name is the same.

### Expected Behavior

I believe that this is counterintuitive and wastes effort. My workflow overwhelmingly is to copy a model and paste it as a new model into another file. If I do want to replace an existing model, I would delete the old model and paste in the new. Or, if they have the same name, then I'd expect to be able to paste over the existing model with the same name, getting an overwrite prompt.

As a reference, I'd expect it to behave more like the Windows File Explorer with copy/paste of files.

### Steps To Reproduce

1. Open/create two model and settings files (A and B)
2. Create a model if necessary
3. Copy a model from one file
4. Select the other file and paste model

### Version

2.7.0

### Transmitter

FrSky X10 / X10S (ACCST)

elecpower, 2022-05-03:
Tested in Windows 10 and Ubuntu 20.04 and working for me.

Drag-n-drop in between destination models creates a new model in the destination file with the name and settings from the source file.

Drag-n-drop onto a model and paste in the destination file both display this message.

[image]

Paste does not allow creating a new model however drag-n-drop between existing models will achieve this or click + Model and paste onto the new model.

Model names do not have to be unique in a model file and the same model name does not guarantee that all settings are the same for the same named models. Therefore why drag-n-drop and copy and paste do not work as say Windows Explorer.

RJMcInty, 2022-05-04:
Thanks for looking at this! I played around with this a bit more, and here's what I see on Win 10 (possibly this is by design, but it's confusing to me if so):

- Drag/drop only works when you:
      - Drop on an existing model: it pops the dialog and replaces the model
      - Drop on a category heading: it creates a new model with the contents of the paste
- Copy/paste:
      - Creates a new model if the destination file is empty
      - Tries to overwrite whatever model is selected if the model file isn't empty
      - But WILL create a new model if you select a category heading before pasting (because in an empty file the category heading is the only thing that can be selected)

The crux of the issue for me is that I'd expect that if I copy from one, and right-click _any_ blank space in the destination model file and hit paste that it wouldn't try to overwrite the currently selected model, rather it would create a new model and paste into it. If I wanted to paste over a model I'd copy and then specifically right-click on the model I wanted to replace (vs. right-clicking in blank space)

I hope I'm not confusing things more; I could try to screen record a video and post it somewhere, but am not really set up to do that.

Thanks for all you do!

elecpower, 2022-05-04:
Drag/drop will also work if you move slowly between the models and a bold line appears. At that point release the mouse. You cannot drop below the last model even though it looks like you can ie a bold line. But you can shuffle after dropping.

Yes paste to the blank area does not work as the blank space is not drop event aware. The details tree/list is what responds to the drop event.

elecpower, 2022-05-04:
@RJMcInty FYI I have updated my previous comment

RJMcInty, 2022-05-08:
Ok, thanks for the discussion. The user experience seems self-consistent, though not quite consistent with things like file explorers. But now that I understand it, I can use it.

One suggestion might be, though, to rework the "Are you sure" dialog to add the option of creating/pasting a new model, in addition to overwriting.

If that idea doesn't have merit, then please feel free to close this (or I can).

elecpower, 2022-05-11:
The current implementation is not ideal so let's leave it open and tag as enhancement

PR #1976 (2022-06-07): Companion option to insert if pasting onto existing model
Fixes #1929

Summary of changes:

[image]
closed 2022-06-07 by pfeerick