EdgeTX/edgetx #1930: Widget Parameter Selection using Crossfire

mrduncan626, 2022-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Widget parameter selection using Crossfire telemetry are limited to only the parameters in the attached pic.  All other sources  selected in the widget results in the same limited parameters.

[image]

### Expected Behavior

To display Crossfire telemetry parameter/values on the radio screen via the widget.  There were no issues in displaying the Crossfire telemetry values in the widget in V2.6.0.

There are no issues displaying FrSky telemetry values in the widget on V2.6.0 or 2.7.0.

### Steps To Reproduce

Widget screens are setup manually through the radio telemetry screen setup.

### Version

2.7.0

### Transmitter

Radiomaster TX16S

### Anything else?

The multi-module firmware is V1.3.3.7.

pfeerick, 2022-05-03:
I'm not quite sure what the problem is... are you saying that you only have those seven options for source, or just that the crossfire telemetry values are missing (but *are* present if you check the model telemetry tab - i.e. you did do sensor discovery)? Or if you mean it keeps jumping to back to 6P, maybe need to do stick calibration again and also do the 6POS as part of the calibration (sticks, sliders, pots and 6POS).

mrduncan626, 2022-05-03:
Recalibrating the sticks, sliders, pots and 6POS seems to have fixtured the problem.  Thanks.

closed 2022-05-03 by mrduncan626