EdgeTX/edgetx #1931: Allow reading/writing of models only, settings only, or models and settings.

RJMcInty, 2022-05-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I've overwritten new calibration data on the radio by old calibration data in the models and settings file. I tend to use the companion for extensive model creation/updates/management, and then write them to the radio. I often have updated settings/calibration on the radio that I haven't downloaded (I may download the models/settings and then work on the models for several sessions, possibly using the radio in the meantime), and then inadvertently overwrite them when uploading the new models.

In general, I'm still coming to grips with the whole model and settings as a group concept, but this has tripped me up a number of times (getting old/stale calibration data on the radio from a file).

### Describe the solution you'd like

I would like to be able to write just models to the radio, rather than models and settings. Ideally, the read/write to/from radio would provide the option to read/write both models and settings, or just one or the other.

elecpower, 2022-05-03:
Related #932

RJMcInty, 2022-05-03:
> Related #932

Related, but definitely different.  (I did see that one when searching before opening... :) )

elecpower, 2022-05-03:
> I would like to be able to write just models to the radio, rather than models and settings. Ideally, the read/write to/from radio would provide the option to read/write both models and settings, or just one or the other.

IMHO reading models and settings should always transfer all configuration to Companion as there are many radio settings that Companion and Simulator rely upon to control how it allows models to be managed and will also trigger incompatibility warnings and offer to convert if the radio settings are for a radio that is not the same as the current Companion radio profile.

Agree there needs to be more flexibility when writing to radio as evidenced by this and other issues raised. Though not keeping the radio and model settings compatible/in sync can create other issues which will potentially make diagnosing those issues more difficult. Would definitely need a big warning message for this one if progresses.

RJMcInty, 2022-05-04:
Thanks for the response! I definitely get that there may be issues that arise if the impacts of possibly reading/writing only the models aren't carefully considered.

It may be that I'm not fully understanding the expected workflow around reading/writing models/settings, and how persistent (long-running vs. short-term) copies of the Companion model/settings files are expected to be. I may hop over into the Discord and have a more effective back and forth conversation there.

I think that if I could imagine any solution to this, it would be for each category of settings (Setup, Global Functions, Trainer, Hardware, Calibration) to be versioned/timestamped on both the Companion and radio, and when you read or writer it checks to see if you're replacing newer data with older, and if so prompts/confirms.