EdgeTX/edgetx #1933: win64 companion crashing when opening simulator

P-I-Engineer, 2022-05-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

open companion
drag/drop etx file
double click on model
hit simulate
simulator opens, companion crashes out a moment later.

does not happen if i created a new file and hit simulate. only happening with my current file.

### Expected Behavior

not crash

### Steps To Reproduce

see above

### Version

2.7.0

### Transmitter

FrSky X9 Lite / Lite S

P-I-Engineer, 2022-05-03:
nevermind, looks like a crappy file, i copy pasted whole thing over to new doc and works now.

P-I-Engineer, 2022-05-03:
still happening

elecpower, 2022-05-03:
Seems to be a Windows 64 issue as not crashing on Ubuntu 20.04

elecpower, 2022-05-03:
@P-I-Engineer so I can test further can you please attach your etx as a zip and advise what Companion radio profile you are using?

P-I-Engineer, 2022-05-03:
x9 lite 
[document1.zip](https://github.com/EdgeTX/edgetx/files/8614454/document1.zip)
s

elecpower, 2022-05-03:
Using your file it crashes Companion, the simulator window and stand alone Simulator in Win64 and Ubuntu 20.04. 

Loading your file from the stand alone Simulator in Ubuntu the last debug messages are:

> 0.31s: YAML model reader
0.31s: opentxStart(0)
0.31s: playFile("/SOUNDS/en/OMP M2/name.wav", flags=0, id=0)
0.31s: luaInit
0.31s: lsScripts 0x7f9ce0013010
The next is a segmentation fault.

When I create a new file and model the simulator does not crash.

P-I-Engineer, 2022-05-04:
i tried deleting the "one" special function that calls out the model name when i change to that mode. it still crashed.

RJMcInty, 2022-05-04:
It seems to be something with bluetoothMode: TELEMETRY in the radio.yml. Setting it to OFF allows the simulator to launch. With it in TELEMETRY there's an access violation. I don't have a debug build/symbols so can't get a stack trace from the debugger.

P-I-Engineer, 2022-05-04:
what is this in radio setting -> setup
i can't change it

[image]

RJMcInty, 2022-05-04:
According to a comment on the OpenTX wiki, it "allows you to put a specific ACCESS registrationID. You can do that for exemple if you want two radio been able to quickly bind the same models". I did try replacing the odd charset characters with one from my working radio.yml and it didn't make a difference.

I started tracing through the bluetooth/telemetry code and got lost, so I'll step back and let the experts weigh in. :)

Was this .etx converted from an .otx? Or created from scratch? If .otx, do you have the original? It's interesting that there's some odd charset characters in it (owner registration id and bluetooth name). Not sure if that's by design or a different bug.

P-I-Engineer, 2022-05-04:
converted from otx, not sure if i still have it

P-I-Engineer, 2022-05-04:
i get this error now as well

[image]

elecpower, 2022-05-04:
I saw the odd looking values and removed to test and confirm they were not causing the crash.
If you have the otx that would be great as that would help check the conversion to etx.

P-I-Engineer, 2022-05-04:
this should be it

[x9lite-s.zip](https://github.com/EdgeTX/edgetx/files/8617677/x9lite-s.zip)

elecpower, 2022-05-04:
Thanks it has standard characters so will run it thru conversions

elecpower, 2022-05-04:
Registration fields converted from otx to etx by Companion 2.7 successful.

P-I-Engineer, 2022-05-07:
any more thoughts on this?

RJMcInty, 2022-05-08:
Did you do your migration from OTX on your radio (by copying over RADIO and MODELS folders to the ETX SD card you prepared, and then letting the radio convert it? Or by downloading to PC via OTX Companion and saving and opening in ETX Companion? (Or some other way)? When I played with it the other day the OTX -> ETX by ETX Companion worked fine (as far as I could see; I'm not an expert).

It definitely seems to be something specific with your ETX file. How it got that way, though, is the question. :)

P-I-Engineer, 2022-05-08:
Let radio convert if I remember correctly. I'm trying to figure out how to swap stuff around to get it to work. I don't care about losing radio settings

RJMcInty, 2022-05-08:
If the "x9lite-s.zip" you shared earlier is a good backup and you don't care about anything on your SD card (or have a spare one), then maybe:

1. Prep the SD card from scratch (back it up if it's your only one!) on your PC
2. Put it back in the TX and go through fresh startup sequence (SD card prep, calibration, etc.)
3. In EdgeTx Companion, open the OTX and convert it, and review everything
4. Download the radio models/settings to EdgeTx
5. Copy the models from the converted OTX to the new model/setting file created when you download the radio settings
6. Use EdgeTx to write the new model/setting file back to the radio

I'm new to EdgeTx/OpenTx though, so there may be a gotcha there, or a better way. But if you've got a spare SD card then it won't hurt to try it. But, assuming that the OTX file that you're starting from is good, then that should work.

P-I-Engineer, 2022-05-08:
ok, i kind of just did that, deleted my models/radio folder, shuffled some stuff around in companion. now it appears to be working.

P-I-Engineer, 2022-05-08:
tested a bunch of times, simulator is no longer crashing.

different issue now. there aren't any extra model numbers showing up when i read models from radio and it's doubling up my not rx numbered rx's.

[image]

elecpower, 2022-05-20:
PR #1932 will likely fix this. It is in nightlies and therefore should be in release 2.8 unless it is withdrawn

gagarinlg, 2023-02-25:
Is this fixed?

gagarinlg, 2023-03-18:
stale

closed 2023-03-18 by gagarinlg