EdgeTX/edgetx #1934: Inputs Move/Paste Loses Input Name

jmxp69, 2022-05-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.7 Main Input Name is lost when using the move function: Move/Paste.

### Expected Behavior

When Copy/Cut [Move]/Paste is used I would expect all available attributes to be included upon paste.

### Steps To Reproduce

2.7 Main

1. Inputs Tab: Create an input and populate Input Name in the edit screen.
2. Highlight Input with Input Name Populated
3. Click jog dial
4. Select Move
5. Move jog dial to new line
6. Click paste
7. Input Name is erased and input reverts to Line Number

### Version

2.7.0

### Transmitter

Radiomaster TX16S

### Anything else?

N/A

pfeerick, 2022-05-05:
In contrast, I would expect Move to bring the name across, but Copy to leave it alone. If I'm copying an input line, I don't want to set up two inputs to have the exact same name - after all, how do I tell them apart afterwards? I think with OTX it was a little more obvious what/why this happened... it also doesn't move the input name, **but** it also left the old name as it was set... since you can't actually move the input row, just it's contents. 

In other words, I propose that Move should move the name of the input along with it's content (when it is a single row input), which is IMO an improvement over OTX behaviour, and leave Copy as it is.

jmxp69, 2022-05-05:
Leaving the name alone on copy would save the obvious step of changing it. Another thought would be to change Move to Cut which follows the convention on pretty much every OS for the last 30 years. 

Cut/Paste = Keep everything
Copy/Paste = Keep most, lose name

In either case I do think move/cut should be an exact copy.

Thanks

PR #1940 (2022-05-08): move input name together with input string
Fixes #1934 

When the input is moved (Move/Paste - Paste Before/Paste After are not affected) the input name is also moved.
closed 2022-05-08 by pfeerick