EdgeTX/edgetx #1936: Max/min "trim reached" message not persistent

Arvycka, 2022-05-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using trim switches, you get a "minimum/maximum trim reached" voice message on reaching either trim limit. 
Then every subsequent trim click in the same direction gives infinite beeps, just like while increasing/decresing trim value, and there is no way to know that the trim limit was reached without looking at the screen or holding the trim switch in the oposite direction until you get back to the middle.

### Expected Behavior

After trim limit was reached, every subsequent click in the same direction should annouce with a voice message that the limit was reached.

### Steps To Reproduce

1. Keep holding trim switch until you hear a message "minimum/maximum trim reached.
2. Push the same trim switch in the same direction again - you get the same "beep" sound as while changing the trim
3. Keep holding the trim switch - it keeps beeping continuosly while holding the switch.

### Version

2.7.0

### Transmitter

Radiomaster TX16S

eshifri, 2022-05-07:
Looks like this is intentional:
https://github.com/EdgeTX/edgetx/blob/26262c4c92fdbcbefaed1ba3ccd71bda01f8e37e/radio/src/opentx.cpp#L1020

RJMcInty, 2022-05-07:
:) This is something I was going to bring up once I got my build/testing issues sorted out. The use of "beepTrim" seem to be inverted (or I'm just missing something).

It gets initialize here as false:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/opentx.cpp#L986

But here where it's actually used it's negated:
https://github.com/EdgeTX/edgetx/blob/main/radio/src/opentx.cpp#L1051

So, everywhere above that it's setting beepTrim=true it's turning OFF the beep, not ON, which seems counter-intuitive to me, but that's the way it is. I was going to ask about possible fixes for this. I can see two:

1. Play the Max/Min sound every time trim is pressed once it reaches a limit
2. Don't play anything after the Max/Min sound, when trying to go past a limit. It would be like the trim button isn't doing anything.

#1 is *probably* easier, looking at the code. But it also seems like it would be more annoying. But it also really gets the point across that you should stop trying to go in that direction. :)

I have a couple of other questions about code near it that I'll ask on discord if necessary.

eshifri, 2022-05-07:
Your links converted perfectly for me. :-)

I think this is the typical "name does not match action" case. :-) If you want to handle this issue, you may want to improve the code readability as well by renaming the variable into "noBeepOnTrim". :-)

Regarding the fixies I do not have strong opinion, but I can provide additional arguments for both variants:
Solution 2 may be confusing since one can always suspect he is pushing the wrong trim: trims may be disabled in the FM scree.
Solution 1 may be especially annoying if one is using trims 5 and 6 (or any other trim) as a button - not as a trim. Current SW just beeps; if it will announce Max/Min reached with every press....
Of course, the proper way to handle it would be to disable trim in the FM screen, but there are people that do not use FMs...

RJMcInty, 2022-05-07:
Thanks! For the link, yours embeds the actual source code inline in the comment with a link to the file. My links work, but don't have the cool inline look.

I got my builds working, mostly, and have fixes both ways. I noticed the "TRIM_REUSED" section just above the code we've been talking about. There's no change there (in either case) so it should continue to function as-is, in either case. You make a good point about Solution 2, and it could be very confusing when you're not looking at the radio. Not good. So, I'm leaning towards the originally proposed solution (Solution 1).

Does anyone else have any feedback? I'll give others a chance for that while I work on some other build system aspects. And, yes, I'll improve the code/variable as well.

eshifri, 2022-05-07:
Please keep in mind that TRIM_REUSED() covers only the case when trim is used for GV adjustment. There are some other ways to use trim buttons.

RJMcInty, 2022-05-07:
Understood.

pfeerick, 2022-05-08:
To get the inline code block, you need to use the permalink instead - which is better to use anyway as if there are any changes to the code in the meantime your line reference will be wrong ;)

[image]

For solution 2, it itself would cause the same issue being raised here - lack of feedback ;) Plus the additional "is the switch doing anything" concern - meaning you have to look down and away from your craft - which is not an option if flying FPV, etc.

For solution 1, it's ok I guess, but I myself have never found this to be a problem... the pitch of the beep changes as the trim moves, so once the pitch stops changing, even if you missed the 'minimum/maximum trim reached' the lack of trim tone change was a giveaway. Another thing to be wary of is that it will be blocking the audio queue if it plays for a couple times back to back, and what happens when you click the button and it's already playing, etc. I guess if it were to be made this way I would probably prefer it to be an option rather than default behaviour.

RJMcInty, 2022-05-08:
Hey, thanks for the tip! Much appreciated. Yes, I agree with the "no feedback" being undesired. I'll put a draft PR out for people to consider. If it's not what people want, then it doesn't go in; either way it's served its purpose for me: learning. :)

pfeerick, 2022-05-09:
Just a FYI @Arvycka... issues usually remain until a PR is merged (and when linked to an issue it will automatically be closed at that time). If a PR is rejected, the issue isn't resolved ;)

Arvycka, 2022-05-09:
Noted, Thanks. :)

closed 2022-07-14 by pfeerick