EdgeTX/edgetx #1938: Max baud rate only at 1.8M on Jumper T-Pro

jtahanlangit, 2022-05-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Cannot set baud rate to max 5.2M unlike in v2.6 it can set to 5.2M baud rate. The max baud is only 1.8M on v2.7 with Jumper T-pro ELRS version.

### Expected Behavior

Max baud rate to 5.2M

### Steps To Reproduce

F

### Version

2.7.0

### Transmitter

Jumper T-Pro

pfeerick, 2022-05-05:
The internal module connection on the Jumper T-Pro is unable to go higher than 1.87M as no DMA possible. It is a hardware design limitation, which the software is correctly following/enforcing now. The external module bay is not limited though.

closed 2022-05-05 by pfeerick

jtahanlangit, 2022-05-05:
Thanks man for the clarification. Much appreciated

pfeerick, 2022-05-05:
No problem. Just sorry I wasn't able to say "hey, it _is_ a bug, and we can fix it..."

jtahanlangit, 2022-05-08:
Hi Pete,

What about on v2.6.0 the baud rate of 5.2M on Jumper T-Pro is available but
does it mean it doesn't work on that rate?

Cheers mate.

Julius

pfeerick, 2022-05-09:
Correct... 2.6 showing it was a bug which as since been fixed.