EdgeTX/edgetx #1939: Add 3POS option for switches SF & SH

AMainRCRacing, 2022-05-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Switches SF & SH only show options for None, Toggle & 2POS

### Describe the solution you'd like

Add 3POS option on switches SF & SH. All the other switches have None, Toggle, 2POS or 3POS. Some people are replacing SF & SH with aftermarket 3-position switches but there is no way to reflect this in the settings.

pfeerick, 2022-05-06:
It's not as simple as that... you can't add a 3POS switch to a board that only has facility for a 2POS... there are three connections needed for 3POS, and 2 for 2POS, so where does the third wire go? 

So, to start off with, what transmitter is this in relation to?

SBWings, 2022-05-06:
RM TX16S. I didn't realize it was a hardware issue. I was asking for someone on a FB group. I didn't replace my switches. I'll let them know it's not possible. Thanks!

SBWings, 2022-05-06:
The person who did this posted on FB: Well after breaking 3 Switches the other day I decided to buy separate switches 12(pkt) from Amazon "3 pin 3 position On/Off/On" De-soldered the old ones and replaced them with the new, with only one issue. Switch SF was 2 position and not 3 but I can live with that for now. The best thing is that the radio doesn't feel so bulky.
Someone else said to go into switch settings and to change it to 3POS. That lead to 9 more comments about SF & SH not having a 3POS option. I don't think anyone knows it's a hardware thing, not simply a coding change.

sasplayers, 2022-05-06:
Thanks SBWings 
I'm the original FB poster
pfeerick the transmitter is "Radiomaster TX16s" - on the PCB for switch "SF" their are 3 pin's for the 2 position switch, when I've checked for continuity each of the 3 pin's corresponds to 3 of the 5 wires as 2 switches on that same PCB share a ground wire.
hope this makes sense.

pfeerick, 2022-05-06:
Yes, unfortunately it doesn't actually go any further... whilst they are wired individually (e.g. two sides of SH are blue and yellow, the 'alternate' position of the 2POS switch is simply connected to ground on the breakout board PCB on they are plugged into... i.e. if you unplug the connector, you'll find out when you measure continuity on the socket, that the top pin (blue wire) is connected to the bottom (black, GND). So they are not actually connected up to the mainboard and thus can't be "upgraded" to 3POS. 

[2.17. Switches SA to SH](https://www.rcgroups.com/forums/showthread.php?3869543-Blog-17-RadioMaster-TX16S-schematic-diagram)
[image]

sasplayers, 2022-05-06:
Oh I do see what you mean, I guess when I checked for continuity the switch was in the off position, which makes me wonder if I cut that trace and redirect it to Pin 1. I have to change another switch today so I'll let you know any further developments.
Thanks again pfeerick

sasplayers, 2022-05-07:
I had a close look at the PCB to see how easy it would be to change from 2 position to 3, and it seems to be easy enough to just move the Blue wire to the 3rd pin on the switch, the only issue is the firmware
[image]
[image]
[image]
.

pfeerick, 2022-05-07:
That isn't the problem... note I said "SOCKET" before, not plug... if you measure the SOCKET on the radio, you'll find that the top and bottom pins are shorted - i.e. the top pin is connected to GND. This top pin also does not connect from the breakout board on the transmitter to the main processor board at all, so there is nothing for the software to monitor! 

The switch PCB doesn't need any modification or re-work at all... for SH blue is connected to one side, and yellow the other. So *IF* the breakout board didn't connect it to GND, and *IF* the breakout board connected that top pin to the mainboard (via the ribbon cable), then 3POS could become an option.

sasplayers, 2022-05-07:
oh  just my luck ; I didn't look at the other PCB, I guess I will change them 2 switches back to 2 position, thanks again for all the help Peter.
I can now atleast fly for the time been (now that I can arm the drones)

pfeerick, 2022-05-07:
Indeed... hard to fly without an working arming switch! 😆

closed 2022-05-07 by pfeerick

rotorman, 2022-05-07:
@sasplayers FYI: TX16S mainboard has SI and SJ auxiliary inputs - in case you are in search for two additional switch inputs.

Even if you would swap SF/SH against 3-Pos and wire the additional poles up to SI and SJ, that would NOT not give you in 3-pos switches in software, but as four 2-pos switches (SF, SH, SI, SJ).
If you really want SF/SH to be 3-pos in EdgeTX (hooked up to SF, SH, SI and SJ ports, but handled in software only as 3-pos SF and SH), you would need to edit a bit the source code and do custom build for yourself.