EdgeTX/edgetx #1942: xlites 2.7.0 no isrm internal module available on X lite PRO

timemaster5, 2022-05-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I've built the latest edgetx 2.7.0 f79978b6 with bluetooth,crossfire, ghost, noheli, internalmulti, internalaccess, multimodule and luac options, but I can't choose an ISRM module in the model settings.

### Expected Behavior

As in the previous versions, I would also like to use the internal ISRM2 module on 2.7.0 :)

### Steps To Reproduce

The only thing is to download the official firmware release and test it on X lite PRO. But to be on the same page, I use these options: cmake -Wno-dev -DPCB=XLITES -DHELI=NO -DLUA=YES -DGVARS=YES -DFONT=SQT5 -DMODULE_PROTOCOL_FLEX=YES -DAFHDS3=NO ../

### Version

2.7.0

### Transmitter

FrSky X-Lite / S / Pro

pfeerick, 2022-05-07:
So just to double check, when you go into `SYS -> Hardware -> Internal RF` ISRM isn't an option there? I don't have that hardware, so am relying on Simulator for initial analysis

https://user-images.githubusercontent.com/5500713/167240838-bd0ef1b4-a7e9-4b56-85cd-578a2f7bc9b9.mp4

timemaster5, 2022-05-07:
All right, stupid me. I didn't come across this option before on this radio. So when I set the ISRM there, all is fine.

So is it really a bug? I was upgrading and saw the updating EEPROM message. Maybe that's why I wasn't expecting that, and not being able to select ACCESS protocol in the model menu gave me the impression that the support is just not there.

raphaelcoeffic, 2022-05-07:
@timemaster5 so what you’re basically saying is that the internal module setting was not correct after upgrade, right?

raphaelcoeffic, 2022-05-07:
By the way, can you please provide the exact command you used with CMake? It might be related to the default module & CMake.

timemaster5, 2022-05-07:
> @timemaster5 so what you’re basically saying is that the internal module setting was not correct after upgrade, right?

yes

> By the way, can you please provide the exact command you used with CMake? It might be related to the default module & CMake.

`cmake -LH ../ > cmake-options.txt
cmake -Wno-dev -DPCB=XLITES -DHELI=NO -DLUA=YES -DGVARS=YES -DFONT=SQT5 -DMODULE_PROTOCOL_FLEX=YES -DAFHDS3=NO ../
make -j`nproc` firmware
`

PR #5236 (2024-07-01): fix: supported internal module for monochrome radios w/ ISRM
Fixes #5234
Fixes #5153

pfeerick: I am going to tentatively say this resolves #1942 also as this gem has been present since 2.6, so could have been responsible since it was a 2.7 issue. 

Summary of changes:
- use `DEFAULT_INTERNAL_MODULE == ISRM_PXX2` as a better heuristic to determine the supported modules.

Note: it seems X9lite and X9liteS are impacted as well.