EdgeTX/edgetx #1950: Please add a method to save and retrieve Companion Telemetry Simulator values

dwheeld, 2022-05-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is very frustrating to have to re-load these values each time I want to test new logical switches or alarms with the telemetry simulator. If I need to go back and edit something, it is again necessary to manually enter the values when returning to the simulator. 

### Describe the solution you'd like

A method to save the values I have entered, and then reload them later would be my choice. Use two buttons as shown in the attached picture. 

### Describe alternatives you've considered

The push buttons, circled in red, could be used to save/reload the values for all of the sensor values.

[image]

pfeerick, 2022-05-11:
How about a method of being able to save it as a SD log file? The load functionality is already there, would just need the save ability?

dwheeld, 2022-05-11:
I don't really understand. It isn't a log, it's just static values that are entered to test alarms, etc. A simple text file with the 31 or so values (for RxBt thru AccZ),  is what I am referring to. The user clicks the save button and enters a name. for them to be saved under. Later, after leaving to edit alarms, they restart the telemetry simulator and read the values back in by clicking the load button and selecting the saved file. 

QTextStream out, and in, is what I would use. I've attached telemetrysimu.cpp; .ui; and .h files modified to perform this. They go in companion\src\simulation. 

[telemetrysimu.zip](https://github.com/EdgeTX/edgetx/files/8665748/telemetrysimu.zip)

pfeerick, 2022-05-11:
And how do you cater for being able to save or load different sets of data?

dwheeld, 2022-05-11:
Save the different sets of data under different file names, test1.txt, test2.txt, mydefaults.txt, etc.

pfeerick, 2022-05-11:
Ah, sorry! I didn't realise you had implemented/considered being able to specify filename. In that case sounds good. I was only raising the log file functionality as all the load code is there (no reinventing wheel required), and single row file would do just the same job + be self-descriptive ;)

dwheeld, 2022-05-11:
I entered a pull request for this feature.

dwheeld, 2022-05-14:
Is anything further expected from me on this?

closed 2022-06-15 by raphaelcoeffic