EdgeTX/edgetx #1952: Setting USB-VCP to OFF is not persistent

mha1, 2022-05-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.7.1: Setting USB-VCP to OFF doesn't survive radio life cycle. USB-VCP is set back to CLI in next life cycle.

This is probably also the reason why it seems as if USB-VCP defaults to CLI while (I think) it should default to OFF, shouldn't it?

### Expected Behavior

USB-VCP set to OFF is stored persistently

### Steps To Reproduce

Go to Radio Settings/Hardware
Set USB-VCP from CLI to OFF
Leave Hardware Tab
Turn off radio
Turn on radio
Go to Radio Settings/Hardware
Observe USB-VCP setting is back to CLI

### Version

2.7.0

### Transmitter

Radiomaster TX16S

### Anything else?

initial radio.yml excerpt with USB-VCP set to CLI
[image]

radio.yml excerpt after setting USB-VCP set to OFF before turning radio off
[image]

radio.yml excerpt after turning radio on again
[image]

pfeerick, 2022-05-11:
Transmitters that support ELRS passthrough (i.e. TX16S and Zorro) will always reset USB-VCP being set to CLI if it is set to OFF, as failing to do so will prevent passthrough updates from functioning, as it depends on CLI being active. This was a deliberate choice, so is by design, not a bug. I guess it could be improved slightly by also checking for whether the internal module is CRSF or not, but a future option being considered is be being able to also do pass-though update for the MPM.

mha1, 2022-05-11:
Thanks for the explanation (which is not obvious :-)). Works for me.

rotorman, 2022-05-11:
I wonder if it might be better to remove USB-VCP Option "OFF" from the menu of radios that support pass-through flashing of internal Express LRS in order to make this more intuitive/clearer.

mha1, 2022-05-11:
more intuitive -> more good :-)

pfeerick, 2022-05-11:
Yeah, that probably is the way to go... yes, it *does* mean if the VCP is set to anything other than CLI that passthrough will fail, but that is the case now anyway, so is more of a user education problem. I think it should be specifically tied to the internal module selected, and is easy to filter out on colorlcd, and B&W isn't that much harder. But companion needs to know about this also.

rotorman, 2022-05-13:
@mha1 could you please test #1957

mha1, 2022-05-13:
I have checked and am not sure if I understood the change correctly. Trying this on my TX16s with internal module set to MULTI. I can't set USB-VCP to CLI anymore. But when I set it to OFF it'll be back to CLI again in the next life cycle.

[image]