EdgeTX/edgetx #1961: Companion crashes when opening model or simulate if SD-Path in application settings not empty

mamaretti, 2022-05-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

when I try to start simulator or try to edit a model Companion crashes (2.5, 2.6, 2.7....)

but only when entered a correct SD-path in App-settings (settings->radio-profile->other settings)

if the field is empty  simulator and edit a model is working.

i'm on macOS Mojave

### Expected Behavior

no crash

### Steps To Reproduce

open document.etx file an try to start simulator.
check sd-path settings before.

### Version

2.6.0

### Transmitter

Radiomaster TX16S

### Anything else?

[image]

mamaretti, 2022-05-13:
solved by myself.
it crashes with custom theme.
i created a new theme and now its working.
similar issue https://github.com/EdgeTX/edgetx/issues/1517#issuecomment-1025404356

elecpower, 2022-05-13:
> it crashes with custom theme.
> i created a new theme and now its working.

@mamaretti I have been trying to replicate and cannot reproduce so I am obviously not taking the same steps. Could you please explain further. Thanks.

mamaretti, 2022-05-16:
hi elepower.
on EdgeTX 2.6 I created a custom theme some time ago.
on the Radio the theme ist working without problems.

companion can read the SD-Data by entering the correct path  to system location.

what I have done: I deleted my theme and  selectedtheme.txt 
so that EdgeTX and companion load the default EdgeTX-Theme. 

after restart companion simulator and modell-editor works again correctly.

then I created I a new custom theme via simulator. and now companion, modell-editor and radio is working with new theme.

I compared the both "theme.yml" files. and the only difference was that the "corrupted one" has a DASH "-" in the info-attribute of summary. perhaps this was the problem.

mamaretti, 2022-05-16:
currupt , old theme (?)

`---
summary:
  name: Blue
  author: Me
  info:

colors:
  PRIMARY1:   0x000000
  PRIMARY2:   0xF8FCF8
  PRIMARY3:   0x0074B0
  SECONDARY1: 0x282C28
  SECONDARY2: 0xC8E8F8
  SECONDARY3: 0x98C0D8
  FOCUS:      0x283880
  EDIT:       0x4074E8
  ACTIVE:     0xE0EC90
  WARNING:    0xE00000
  DISABLED:   0x888C88`

ok:

`---
summary:
  name: blue2
  author: me
  info: 

colors:
  PRIMARY1:   0x000000
  PRIMARY2:   0xF8FCF8
  PRIMARY3:   0x0074B0
  SECONDARY1: 0x282C28
  SECONDARY2: 0xC8E8F8
  SECONDARY3: 0x98C0D8
  FOCUS:      0x283880
  EDIT:       0x4074E8
  ACTIVE:     0xE0EC90
  WARNING:    0xE00000
  DISABLED:   0x888C88
`

elecpower, 2022-05-16:
@mamaretti thank you for the info. The dash shouldn't crash Companion  but hey stranger things have happened. I'll experiment so more.

mamaretti, 2022-05-16:
I did a file compare an the only difference was the dash and the name,
if you want I can send you my SD-Backups

elecpower, 2022-05-16:
The dash may not be the dash character as it does not display in the email copy of your post. Could you please upload the suspect theme.yml file as a zip file so I can check the file in a hex editor

mamaretti, 2022-05-16:
attached the hole theme-folder with the non-working-theme
[THEMES.zip](https://github.com/EdgeTX/edgetx/files/8698140/THEMES.zip)

elecpower, 2022-05-16:
Your theme generates the following

> terminate called after throwing an instance of 'YAML::ParserException'
  what():  yaml-cpp: error at line 5, column 9: illegal block entry

Confirms cause is the dash/hyphen. If the hyphen is enclosed in quotes then no crash as treated as text.

elecpower, 2022-05-16:
@mamaretti thank you for your assistance

ZZ-Cat, 2022-05-31:
I have experienced something similar to this.
I first noticed it in Companion 2.7.1.
I rolled back all the way to 2.6.0 & did a completely fresh install of Companion 2.6.0.
I left the SD Structure Path & Backup Folder fields blank; loaded up a new file; created a new model; & everything worked fine.

The very second I filled out the SD Structure Path & Backup Folder fields with where I normally put them (in my One Drive), sure enough... opening the same blank model crashed Companion.

The interesting part is... I recently updated Windows 10 earlier in the week. Before that update, Companion has _never_ crashed on me. Not even once. Perhaps the new update may have changed the way file access permissions are handled? 🤔

PR #1970 (2022-06-04): Stop Companion crashing due to malformed theme.yml files
Fixes #1961
Fixes #1517

Summary of changes:
- trap and report YAML parsing errors when processing theme.yml files
- continuing loading model

Example
[image]
closed 2022-06-04 by pfeerick